FreeCAD/FreeCAD #15004: Neumann style boundary condition (Electric Flux) missing from electrostatic potential constraint (FEM / Elmer)

akapelrud, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is currently no way of specifying neumann style boundary conditions, i.e. the electric flux / surface charge density, in the electrostatic potential constraint.

- An electric flux unit needs to be added.
- The task panel for the constraint must be updated with the electric flux and a selection/radio box (diriclet | neumann)
- The elmer writer must be altered to write the electric flux boundary conditions to the `.sif` file.

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Norwegian Bokmal/Norway (nb_NO)
[/code]
```

### Subproject(s) affected?

FEM

### Anything else?

https://forum.freecad.org/viewtopic.php?p=759498&sid=f01dee1d15afb7a4869da6027617e74a#p759498

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
@FEA-eng FYI

luzpaz, 2025-01-18:
Hi @akapelrud just checking in that you're cool with the forking of your contribution?

akapelrud, 2025-01-18:
Yeah, go for it!

closed 2025-01-27 by chennes