FreeCAD/FreeCAD #15009: CAM Wb: Unable to change cut direction on a profile cut.

ShamanTcler, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In a profile toolpath tried to change from Climb to Conventional. 

Report view:
06:59:13  Traceback (most recent call last):
06:59:13    File "C:\FreeCadBeta\Mod\CAM\Path\Op\Gui\Base.py", line 251, in pageGetFields
06:59:13      self.getFields(self.obj)
06:59:13    File "C:\FreeCadBeta\Mod\CAM\Path\Op\Gui\Profile.py", line 78, in getFields
06:59:13      obj.Direction = str(self.form.direction.currentData())
06:59:13      ^^^^^^^^^^^^^
06:59:13  ValueError: {'sclassname': 'class Base::ValueError', 'sErrMsg': "'Conventional' is not part of the enumeration in Back#Profile001.Direction", 'sfile': 'C:\\bld\\freecad_1719273226893\\work\\src\\App\\PropertyStandard.cpp', 'iline': 466, 'sfunction': 'void __cdecl App::PropertyEnumeration::setPyObject(struct _object *)', 'swhat': "'Conventional' is not part of the enumeration in Back#Profile001.Direction", 'btranslatable': False, 'breported': False}

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * btl 0.9.9
  * Curves 0.6.39
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ShamanTcler, 2024-06-28:
[Back.zip](https://github.com/user-attachments/files/16032297/Back.zip)

This is a back to a Craftsman style coat rack. It has two toolpaths, Rough Profile and Finish Profile, changing either one cause the issue.

It is an enum situation ....

jamwaffles, 2024-06-29:
This was probably caused by me in #14364 - apologies. I'll see if I can reproduce this issue and open a PR with a fix.

jamwaffles, 2024-06-29:
I think this is because the previous variant names `"CW"`, `"CCW"` are present in the uploaded repro file and are loaded verbatim into FreeCAD, so when the code using the new values (climb/conventional) tries to do some matching, it falls over and prints the error you posted.

I doubt it would be too difficult to convert the values on file open (or support both sets of names when loading) but seeing as we're pre-1.0 (just), I'm not sure what the policy here is in terms of supporting existing files.

ShamanTcler, 2024-06-29:
Are you suggesting I as a user make changes?

jamwaffles, 2024-06-29:
Ah, not at all! Sorry if my comment came across that way. I was pretty much asking other FreeCAD contributors whether we should convert/support files using the previous terminology or not.

Syres916, 2024-06-29:
@jamwaffles purely as an example, in the back of mind I had an idea that the Draft/Arch devs had changed some attributes in the past and still enabled previous files to be opened but they appear to have used a new name for the attribute, see https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Draft/draftobjects/patharray.py#L413-L431

knipknap, 2024-07-28:
@jamwaffles If old files are not automatically translated from the previous values, it would be good to at least let the user edit/update the direction. As of now, it seems that doesn't work; the property editor does not offer the new values in the enum, so the operation cannot be restored.

jbaehr, 2024-08-25:
> I was pretty much asking other FreeCAD contributors whether we should convert/support files using the previous terminology or not.

In my eyes: absolutely yes! I consider breaking existing profile and helix operations this way is a release-blocking blocking defect.

maxwxyz, 2024-11-11:
fixed by https://github.com/FreeCAD/FreeCAD/pull/17655

PR #17655 (2024-11-11): CAM: Fix CW/CCW to Climb/Conventional rename
With #14364 a fairly large search/replace PR was merged in order to address #14314.
Unfortunately, this resulted in quite some collateral damage (e.g. #15643 and #15009) and misunderstandings (e.g. #15206 -- just renaming things does not mean the feature got implemented).

With this PR I try to fix the above mentioned bugs, address the critics on #14364 from @Connor9220 and @jffmichi and actually implement #14314.

It's still incomplete and I know that it's quite late for v1.0, but let's hope for the best. I just don't want to see a v1.0 released th [... cut]
closed 2024-11-11 by maxwxyz