FreeCAD/FreeCAD #15013: FEM: Manual range specification for the color legend stopped working

FEA-eng, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed the following regression (it works in 0.21.2) - it's not possible to manually specify the range for the color legend anymore - just nothing happens and the values are reset.

0.21.2:
[image]

0.22:
[image]

@marioalexis84 FYI

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-30:
Confirmed.

But it only happens for FEM stuff while for the curvature plot of a surface mesh (mesh workbench) it still works correctly.

wwmayer, 2024-06-30:
The regression is caused by https://github.com/FreeCAD/FreeCAD/pull/13820 because in ViewProviderFemPostObject::OnChange() true is passed to WriteColorData to always override the value range.

FEA-eng, 2024-06-30:
> The regression is caused by #13820 because in ViewProviderFemPostObject::OnChange() true is passed to WriteColorData to always override the value range.

@marioalexis84 Could you have a look at this?

marioalexis84, 2024-06-30:
@wwmayer  thanks for noticing the regression.

PR #15476 (2024-07-21): Fem: Fix manual color bar range specification - fixes #15013
Fixes #15013

@FEA-eng
commit f466f087 (2024-07-21): Fem: Fix manual color bar range specification - fixes #15013
closed 2024-07-21 by chennes