FreeCAD/FreeCAD #15014: robot positioning not fixed in spite of assigning a location

roli-poli, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I crate a Fcstd file, or one can open the robot sample that is included with FreeCAD: RobotExample.FCStd
Add a second robot and change its placement. Save the file as a new file, close it, and then re-open it.
The second ore all robots that are loaded in the Robot Workbench  are suddenly at the same spot as the first robot, although the placement property is different.
To get you good placement back you need to select any of the robots and run "Edit -> Transform" in the menu (or "Edit -> Placement -> apply -> ok. Then the robots jump to the previously  allocated position 

the bug was reproduced in FreeCAD 0.22.0dev. and confirmed in the FC forum and 

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.5
  * BillOfMaterials 0.0.3
  * btl 0.9.9
  * CADExchanger (Disabled)
  * cadquery_module
  * Defeaturing 1.2.2
  * dxf-library
  * DynamicData 2.60.0
  * Estimate 0.1.2
  * ExplodedAssembly
  * FeedsAndSpeeds 0.5.0
  * InventorLoader 1.4.0
  * Marz 0.0.32-alpha
  * Movie 2023.6.10
  * pluginloader
  * QuickMeasure 2022.10.28
  * Render 2023.8.18
  * sheetmetal 0.3.1
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

[image]
[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
could you test if this bug persists in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

roli-poli, 2024-06-30:
yes the bug is still persisting 100 % same behaviour 

OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * A2plus 0.4.61
  * Assembly3 0.12.0
  * Assembly4 0.50.5
  * BillOfMaterials 0.0.3
  * btl 0.9.9
  * CADExchanger (Disabled)
  * cadquery_module
  * Defeaturing 1.2.2
  * dxf-library
  * DynamicData 2.60.0
  * Estimate 0.1.2
  * ExplodedAssembly
  * FeedsAndSpeeds 0.5.0
  * InventorLoader 1.4.0
  * Marz 0.0.32-alpha
  * Movie 2023.6.10
  * pluginloader
  * QuickMeasure 2022.10.28
  * Render 2023.8.18
  * sheetmetal 0.3.1

closed 2025-07-22 by github-actions[bot]