FreeCAD/FreeCAD #15015: No module named 'PySide.QtWidgets'

AlemSnyder, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was using the new Render addon/workbench and it give the error "No module named 'PySide.QtWidgets'".

In the python console I get:

```Python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',20)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> import PySide
>>> import PySide.QtWidgets
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'PySide.QtWidgets'
>>> from PySide import QtWidgets
Traceback (most recent call last):
  File "<input>", line 1, in <module>
ImportError: cannot import name 'QtWidgets' from 'PySide' (/usr/share/freecad/Ext/PySide/__init__.py)
>>> from PySide2 import QtWidgets
>>> import PySide2.QtWidgets
>>> 
```

### Full version info

```shell
FreeCAD version 0.20.2 (debian apt package)
Render version 2024.6.5
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

I fixed the problem by adding the file `QtWidgets.py` with the contents `from PySide2.QtWidgets import *` to `/usr/share/freecad/Ext/PySide/`. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
Please create the issue here: https://github.com/FreeCAD/FreeCAD-render/issues

closed 2024-06-28 by maxwxyz

AlemSnyder, 2024-06-28:
I [did](https://github.com/FreeCAD/FreeCAD-render/issues/415). It's not an issue with Render, but with FreeCAD 20.2.
I would find the file, but the PySide file is generated in some way I don't understand.

maxwxyz, 2024-06-29:
Does is work in the latest 0.22dev version and the 0.21.2 stable?

AlemSnyder, 2024-07-08:
I know that there is no issue on 0.21.