FreeCAD/FreeCAD #15016: Cannot load models or create new ones

iplayfast, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

built a fresh pull and June 28,2024. Started FreeCAD and it looked fine. 
Created a new file and it hung. On the command line there was the following output
```
FreeCAD 0.22.0, Libs: 0.22.0devR37926 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

connect failed: No such file or directory
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: QOpenGLContext creation failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
... repeating...
```
FreeCAD was hung at that point.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37926 (Git)
Build type: Unknown
Branch: main
Hash: 01874ca898cb5518d4ee2ad5560309ea24be258e
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

iplayfast, 2024-06-28:
I remembered I upgraded nvidia drivers and hadn't rebooted. After rebooting it seems to work. Closing this issue. Operator error.

closed 2024-06-28 by iplayfast

SteveF48, 2025-03-20:
1.0.0 displays the same error whenever I try to access/start projects. It currently displays freecad error (qt.qpa.backingstore) composeAndFlush: MakeCurrent() failed (648 times)