FreeCAD/FreeCAD #15017: Snap to midpoint does not work if near snap is active

kaiwas, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Demonstration not video.

https://github.com/FreeCAD/FreeCAD/assets/32273833/dbd2aa1d-b970-4d70-b465-6a8e16ba0e7b

The current behavior may be normal, but I think it can be improved.
When snapping to vertices, middle and near is enabled they should all work, but snapping to top and middle should have priority.
Now, in order to stick to the middle, you need to disable the snap near.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-07-24:
This is a known issue. If END, MID and NEAR snap are active and an edge is hovered, 4 snap points are found: 1xMID, 2xEND and 1xNEAR. The point that is nearest to the current cursor position wins. In most cases this is the NEAR point obviously. It would be possible to prioritize END and MID snap, but this means that NEAR snap will never win and effectively stop working.

kaiwas, 2024-10-18:
It looks like there was a problem with snapping to the center of the circle.

https://github.com/user-attachments/assets/ac48eb57-aeec-4f31-a678-8ea9a9315756

closed 2025-03-11 by yorikvanhavre

kaiwas, 2025-03-11:
Thank you, this is a great change!

Is it possible to configure the range of the snap somewhere?
The current behavior when the snap is triggered halfway across the screen does not seem like a very convenient solution.
I can open an issue on this topic if necessary.

https://github.com/user-attachments/assets/b58ef317-2c0a-4ff4-bbce-f7be893877f9

PR #20118 (2025-03-11): Draft: Correct Near snap override behavior
Fixes #15017.

Currently Near snap overrides all other 'on-object' snaps such as Midpoint snap and Endpoint snap. With this PR Near snap no longer 'wins' if a different snap point is within snapRange of the cursor.