FreeCAD/FreeCAD #15018: PartDesign: Polar pattern doesn't work on subtractive helix

FEA-eng, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are a few issues regarding the subtractive helix reported here but this one is about a polar pattern applied to it so might be a different cause. It's not a regression since it was initially discovered in 0.21.2 by a YouTuber.

Forum thread: https://forum.freecad.org/viewtopic.php?t=88726

[Helix pattern.zip](https://github.com/user-attachments/files/16033254/Helix.pattern.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
@FlachyJoe FYI

FlachyJoe, 2024-06-28:
An edge is missing
[image]
I saw it previously with helix. OCC fails to ends the shape, it's not due to FreeCAD and certainly requires a deep search in OCC code.

luzpaz, 2024-07-27:
Clarifying, this is an OCC bug ?

FEA-eng, 2024-07-27:
> Clarifying, this is an OCC bug ?

Apparently yes.

AIRCAP, 2024-10-18:
funny workaround: if you do not refine the subtractive helix, the polar pattern works (polar pattern can be refined)

could using refined substractive shapes (which basically get completely merged into the base body) are used in a pattern or is this just by chance?

wolflu05, 2024-12-19:
Thank you so much, this workaround has helped me a lot 🎉 . Was already thinking of using a different, more complicated way than helix to archive my needs. (This was the first time I tried out the helix tool actually). Hopefully this gets fixed so that nobody else runs in to the same problem. (For me I only saw the issues when exporting as stl and tried to slice)

xpol, 2025-03-12:
Same here with FreeCAD 1.0 on MacOS.

```
OS: macOS 15.3.1
Architecture: arm64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Manipulator 1.5.7
```

[image]