FreeCAD/FreeCAD #15019: Error importing an svg

riopar10, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Error importing an svg
18:07:41  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\importSVG.py", line 1793, in insert
    parser.parse(pythonopen(filename))
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\expatreader.py", line 111, in parse
    xmlreader.IncrementalParser.parse(self, source)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\xmlreader.py", line 125, in parse
    self.feed(buffer)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\expatreader.py", line 221, in feed
    self._err_handler.fatalError(exc)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\handler.py", line 38, in fatalError
    raise exception
<class 'xml.sax._exceptions.SAXParseException'>: C:/Users/Antonio/Downloads/Sin título-1.svg:1:2: not well-formed (invalid token)
18:09:26  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.21\Mod\Draft\importSVG.py", line 1793, in insert
    parser.parse(pythonopen(filename))
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\expatreader.py", line 111, in parse
    xmlreader.IncrementalParser.parse(self, source)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\xmlreader.py", line 125, in parse
    self.feed(buffer)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\expatreader.py", line 221, in feed
    self._err_handler.fatalError(exc)
  File "C:\Program Files\FreeCAD 0.21\bin\lib\xml\sax\handler.py", line 38, in fatalError
    raise exception
<class 'xml.sax._exceptions.SAXParseException'>: C:/xxxx/xxxxx/xxxxx/Sin título-1.svg:1:2: not well-formed (invalid token)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Spanish/Spain (es_ES)
Installed mods: 
  * A2plus 0.4.61
  * BIM 2021.12.0
  * fasteners 0.4.72
[/code]
```

### Subproject(s) affected?

Part

### Anything else?

no

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
could you test it if there is still an error with the latest dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-07-02:
@riopar10 can you retest on latest dev and also attach SVG to this ticket ?

riopar10, 2024-07-03:
ERROR SOLUCIONADO.
Despues de un par de días sin solución, en un foro me dijerón que podia ser de corel, y es correcto, el motivo era que cuando guardaba, no se por qué, la codificación no estaba bien, la puse a utf8 y a funcional correctamente.
gracias

closed 2024-07-03 by riopar10