FreeCAD/FreeCAD #15023: Spreadsheet delete key doesn't clear contents in selected cells

hooperbloob, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hitting the delete key when one or more cells are selected will normally clear their contents. 
This doesn't seem to be happening at the moment.

### Full version info

```shell
OS: macOS 12.7.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.64
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
I cannot reproduce on Windows and 
```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.26.1
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.20
  * Curves 0.6.40
```

JohnOCFII, 2024-06-30:
I can confirm.  Would not surprise me if this is a Macintosh specific issue.  

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.6.22
```

wwmayer, 2024-06-30:
The problem is the implementation of SheetTableView::event(). It compares the value of the key event with `Qt::Key_Delete` and this works fine for Windows and Linux but not for macOS because it defines a different [shortcut](https://doc.qt.io/qt-6/qkeysequence.html#standard-shortcuts) (Forward Delete or Meta + D) 

So, the platform-independent implementation should be:
```
        if (kevent->match(QKeySequence::Delete)) {
            deleteSelection();
        }
```

wwmayer, 2024-06-30:
Please test: https://github.com/FreeCAD/FreeCAD/pull/15057

hooperbloob, 2024-06-30:
This looks like the same problem in the Sketcher:  [#15068](https://github.com/FreeCAD/FreeCAD/issues/15068)

PR #15057 (2024-07-24): Spreadsheet: Delete key doesn't clear contents in selected cells
Fixes #15023. On macOS a different shortcut is defined for the 'Delete' key
commit 703c349a (2024-07-24): Spreadsheet: Delete key doesn't clear contents in selected cells

Fixes #15023. On macOS a different shortcut is defined for the 'Delete' key
closed 2024-07-24 by chennes