FreeCAD/FreeCAD #15025: Constraining distance of a reference mode sketch crashes the app

jshet, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

- Open Part Design workbench
- Create a Body
- Create a Sketch
- Switch from "Driving" to "Reference" mode
- Create a rectangle
- Attempt to add a distance constraint to the rectangle (vertical or horizontal)

Expected outcome:
- App crashes

### Full version info

```shell
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * FreeCAD_SheetMetal 0.4.17
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
I cannot reproduce. Could you test this in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

xtemp09, 2024-06-29:
@jshet, please reproduce the crash with the latest weekly build and upload the crash report.

luzpaz, 2024-07-02:
ping @jshet

maxwxyz, 2024-07-07:
I'm closing this for no feedback. You can reply with more and we can reopen.

closed 2024-07-07 by maxwxyz