FreeCAD/FreeCAD #15026: Shortcuts don't work temporarily with Light theme

MiqCG, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When in Light theme, shortcut for new file (Control + n) and others don't work. If new file is created throw File menu with the mouse, then they work (see indications in the bottom part of the screencast).
I think there is the same problem with revision 37841 but apparently not in 37543.
In some situations, clicking with the mouse on the menu bar doesn't work either.
On the other hand, looks like charging the Light theme is much much slower than Classic theme, as you can see in the attached video.

I have searched in Github for [is:issue light theme shortcut] and [is:issue light theme new file] with no relevant results

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.20
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.24
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/c728a522-ab91-4f0b-9fd5-fb72ff43206a

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
@MisterMakerNL FYI

MiqCG, 2024-06-29:
I thought it was not happening in revision 37543 but it is. I edited the first post.
And is happening with "Dark behave" and "Dark theme" themes too.

MiqCG, 2024-06-29:
Same behaviour in revision 37298.
Maybe it's not that keyboard shortcuts are not working but the times/speed of loading of this themes that prevent FreeCAD to obey. 
When you confirm me this hypothesis, I can change the title of the issue.

Loading_times_37928:
https://github.com/FreeCAD/FreeCAD/assets/18265355/95a159b2-3054-4268-9e1f-b90bf6e61725

Loading_times_37898:
https://github.com/FreeCAD/FreeCAD/assets/18265355/c6efd49e-e346-4e00-9b03-8ad82c600809

maxwxyz, 2024-06-29:
@kadet1090 FYI

MisterMakerNL, 2024-06-29:
I cannot reproduce this in windows with my own build.

Syres916, 2024-06-29:
For me this needs to be proven whether it's a distro specific issue, an actual FreeCAD code (which is now looking less likely) issue or a Conda packaging issue. As the OP's distro is originally based on Bookworm and I already have a Bookworm VM, I'll try that first.

If needed I'll then build and attempt to compile on a VM of _Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)_.

Syres916, 2024-06-29:
Could not reproduce the reported issue using Bookworm:

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37931 (Git)
Build type: Release
Branch: main
Hash: f28394a376c285725abdacee3d72bd57a8b196aa
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * CfdOF 1.25.16
  * Assembly4 0.50.12
```

Syres916, 2024-06-29:
Can reproduce the reported issue using AppImage on Bookworm, going from _Dark behave_ to _Light theme_ is like watching paint dry and I can get it to raise FreeCAD 0.22dev is not responding Force Quit/Wait messagebox but the other way round is acceptable.

```
OS: Debian GNU/Linux 12 (bookworm) (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * sheetmetal 0.4.13
  * CfdOF 1.25.16
  * Assembly4 0.50.12

```

I have an older AppImage build 37302 which I'll have to extract and change the stylesheets over but that still won't prove if it's a Conda dependency that's changed, it really needs someone who can compile on Conda IMHO.

MisterMakerNL, 2024-06-30:
Last lag issues where cause by switching Qstyle. But I also see issues with pixmap generation.
And third the biggest difference between behave and others is probably the use of custom colors not sure if those cause extended load times.

Syres916, 2024-06-30:
@MiqCG could you please change the title of the issue to specifically mention applies to the AppImage specifically.

@MisterMakerNL with the AppImage extracted it's absolutely acceptable so it doesn't seem to be a code or stylesheet issue, it's only with the AppImage in it's original state that the time to change is six to seven times longer.
Post release will probably need some more thought as to whether like the Templates for Techdraw they are copied out to the user's Shared folder on the first run or some other left field solution.

Syres916, 2024-06-30:
Absolutely fine, thanks.

kinghat, 2024-07-02:
ive been having terrible freezes with the appimage weeklies for at least a couple months now. here you can see it using only the classic theme selection from start after deleting the freecad `.cache`/`.config`/`share` dirs. so it has a fresh state. it also took like 15s to start(after the loading screen):

[Screencast_20240702_144457.webm](https://github.com/FreeCAD/FreeCAD/assets/16073360/aa6a2726-5462-4ba9-b6d6-9454ad053d00)

> OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.38001 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)

> Operating System: Fedora Linux 40
> KDE Plasma Version: 6.1.1
> KDE Frameworks Version: 6.3.0
> Qt Version: 6.7.1
> Kernel Version: 6.9.6-200.fc40.x86_64 (64-bit)
> Graphics Platform: Wayland
> Processors: 4 × Intel® Core™ i7-5600U CPU @ 2.60GHz
> Memory: 11.6 GiB of RAM
> Graphics Processor: Mesa Intel® HD Graphics 5500

JigglyBuff-Kun, 2024-07-13:
I have the same issue mentioned as above. UI freezes for x seconds for every operation that is being done in FC. Sometimes It works, and I am able to do two operations. But sometimes just clicking on Create Parametric part in the start page freezes the UI for a good amount of time.

OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

Operating System: Arch Linux 
KDE Plasma Version: 6.1.2
KDE Frameworks Version: 6.3.0
Qt Version: 6.7.2
Kernel Version: 6.9.8-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 24 × 12th Gen Intel® Core™ i9-12900K
Memory: 31.1 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 770

JigglyBuff-Kun, 2024-07-13:
I have the same issue mentioned as above. UI freezes for x seconds for every operation that is being done in FC. Sometimes It works, and I am able to do two operations. But sometimes just clicking on Create Parametric part in the start page freezes the UI for a good amount of time.
FC starts with a warning `(qt.qpa.plugin) Could not find the Qt platform plugin "wayland" in ""`
The issue goes way when I use X. This is Wayland specific I guess.
```
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38079 (Git) AppImage
Build type: Release
Branch: main
Hash: de508e437d13e7eb35baffceea496fb9107888cb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)

Operating System: Arch Linux 
KDE Plasma Version: 6.1.2
KDE Frameworks Version: 6.3.0
Qt Version: 6.7.2
Kernel Version: 6.9.8-arch1-1 (64-bit)
Graphics Platform: Wayland
Processors: 24 × 12th Gen Intel® Core™ i9-12900K
Memory: 31.1 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 770
```

MisterMakerNL, 2024-07-22:
should be fixed

kinghat, 2024-07-22:
> should be fixed

then the appimage freezes im having are unrelated as they continue as of:

> 
> OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
> Word size of FreeCAD: 64-bit
> Version: 0.22.0dev.38261 (Git) AppImage
> Build type: Release
> Branch: main
> Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
> Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
> Locale: English/United States (en_US)
> Installed mods: 
>   * OpenTheme 2024.7.3
>   * fasteners 0.5.24
>   * lattice2 1.0.0

MisterMakerNL, 2024-07-30:
I btw see you have opentheme, there was a bug in opentheme where the qtstyle script kept doing it thin causing lag.
Have you updated Opentheme?

kinghat, 2024-07-30:
> I btw see you have opentheme, there was a bug in opentheme where the qtstyle script kept doing it thing causing lag. Have you updated Opentheme?

in freecad ive been running the most up 2 date of everything. most recent addons/themes/dev builds. also fedora which is basically the most up 2 date of everything kde as well.

luzpaz, 2024-07-30:
Is this specific to KDE ?

MiqCG, 2024-07-31:
Keeps happening with revision 38334 in Linux Devuan XFCE desktop.
Classic loads the faster, in 2-4 seconds aprox
Dark (from Classic) loads in 9-22 s aprox
Light (from Dark) loads in 23-27 s aprox

luzpaz, 2024-08-01:
@MiqCG can you please add the full About info to your post? TIA
You are using XFCE and not KDE, correct ?

MiqCG, 2024-08-05:
Same About has in the inicial post

OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git) AppImage
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)

JigglyBuff-Kun, 2024-08-14:
This is fixed when you run as X on KDE. This is KDE issue on wayland.

kinghat, 2024-08-14:
> This is fixed when you run as X on KDE. This is KDE issue on wayland.

this is a freecad appimage(afaik as theres no dev flatpak builds) issue on kde wayland. none of my other apps have this problem.

JigglyBuff-Kun, 2024-08-14:
Yeah, that's true. This is why need beta flatpak builds. I am running X to get FC working.

chennes, 2024-08-31:
@maxwxyz should this get transferred to the Bundle repo? It sound from the above discussion like it's a purely appimage problem, with pretty specific parameters. I'd also suggest that on that basis it's not a release blocker.