FreeCAD/FreeCAD #15028: There are no curvature analysis tools

pierreporte, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are designs that need precise curvature radius and a definite level of continuity: contact (G0), tangent (G1), and curvature (G2). Curvature analysis help designer to meet requirements. The two main tools are:

**Displaying the local curvature radius** on a solid, surface, or line. Solids and surfaces are shaded in a similar fashion as FEM results, and each color corresponds to a precise radius. Lines have a comb similar to splines in sketches. Surfaces can also have a set of combs on top of the shading. In any case, users can pick a point on the geometry to display the curvature radius at this point. Below are some examples from T-FLEX documentation.

[image]

[image]

**Zebra analysis**, which helps identifying discontinuities on solids or surfaces. The geometry is shaded in black and white stripes which width varies with the curvature (radius and orientation). Along the edge, stripes on different faces meet in a different way depending on continuity (G0, G1, G2). The [AutoCAD documentation](https://help.autodesk.com/view/ACD/2024/ENU/?guid=GUID-82E5989F-C943-49A6-A6C6-834B81CC203B) describes how it should look. The relevant section is copied below. The image after the table is another illustration.

| Type | Name | Description | Image |
| ---- | ---- | ----------- | ----- |
| G0 | Position | The position of the surface edges is collocated; they touch. But the tangency and curvature do not match. The zebra stripes do not line up. | [image] |
| G1 | Tangency | The position and tangency of surfaces is the same. This indicates G1 (G0 + G1 or position + tangency). The zebra stripes line up, but they veer away from one another at sharp curves. | [image] |
| G2 | Curvature | The position, tangency, and curvature of the surface edges is the same. This indicates G2 (G0 + G1 + G2 or position + tangency + curvature). The stripes line up, but they do not veer away from each other at sharp curves (because they share the same curvature). This distinction is subtle and a little harder to discern from G1 continuity. | [image] |

[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

The [Curves workbench](https://github.com/tomate44/CurvesWB) by @tomate44 has the `ZebraTool`.

These tools could populate the inspection workbench.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-26:
@pierreporte not the same but maybe related on how this is implemented: the ability to have a draft analysis feature

pierreporte, 2024-08-26:
@maxwxyz I did plan to create an issue for draft analysis but was lazy about this.

maxwxyz, 2024-08-26:
@pierreporte I can do one tomorrow and some other features

pierreporte, 2024-08-26:
@maxwxyz Feel free to do so. I’ll comment on them if needed.

tomate44, 2024-08-26:
For reference, I made a shader-based Draft Analysis tool in Curves WB last year : 
https://forum.freecad.org/viewtopic.php?t=71500&start=50#p694840