FreeCAD/FreeCAD #15030: Inserting column before merged cells can break things

krushia, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this first screenshot, see that column M has a bunch of cells merged.
[image]

Now a new column is created by right-clicking L -> Insert one column(s) left.
[image]

Column M now has the values in weird "pending" state in the cells that used to be merged.

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.40
  * A2plus 0.4.65 (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.13 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.24
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.6.22
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.7.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

Bonus fact:
Oddly they can be recomputed by right-click one at a time, but not by recomputing the spreadsheet. Yet another manifestation of the often-reported recompute bug I suppose.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-06-29:
Looks like this problem is related to this
https://github.com/FreeCAD/FreeCAD/issues/14672

luzpaz, 2024-08-02:
This looks like a dupe of #14672

closed 2025-01-25 by luzpaz