FreeCAD/FreeCAD #15031: Some parameter types disregard Status:Hidden

krushia, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following parameter types seem to get hidden from the property view without Status:Hidden bit being set.
```
App::PropertyMap
App::PropertyMaterial
App::PropertyPythonObject
App::PropertyUUID
Materials::PropertyMaterial
Part::PropertyFilletEdges
Part::PropertyGeometryList
Part::PropertyPartShape
Part::PropertyShapeCache
Part::PropertyShapeHistory
Part::PropertyTopoShapeList
Spreadsheet::PropertyColumnWidths
Spreadsheet::PropertyRowHeights
Spreadsheet::PropertySheet
```
This is creepy and weird. Can they act like all the others please?

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.40
  * A2plus 0.4.65 (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.13 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.24
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.6.22
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.7.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Core

### Anything else?

I put every parameter type in a VarSet just for fun...
[ParameterMuseum.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16038529/ParameterMuseum.FCStd.delete.this.extension.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
@pieterhijma FYI

krushia, 2024-06-29:
Note that my list is from a visual inspection. I just added some missing items, but there could be more because I'm cross-eyed.

I also didn't include sub-properties (is that the right term?) that are hidden -- like `App::PropertyEnumeration`'s `.Enum` -- because I'm not sure if they have a separate status or use the parent's status.

wwmayer, 2024-06-30:
This is not a bug but a missing implementation of an editor of the property types. That's why they don't appear in the property editor view.

However, most of these property types are for internal use only and there it doesn't make sense to implement and editor.

krushia, 2024-06-30:
@wwmayer Thanks for clarifying a common factor. This isn't about the lack of an editor, but that the Property view shows a menu with Status bits (including Hidden) that don't match the observed behavior of the property. I can toggle that Hidden bit on and off again and it has no effect on these types.

pieterhijma, 2024-07-23:
Sorry @krushia, I looked at this more closely and only now I think I understand what you mean. Is it right that the problem is that, for example, PropertyBoolList is hidden by default in the property view, yet it has a status checkbox that you can mark as being hidden?

If so, what would be a proper solution to this? Change the status menu to make sure "[ ] Hidden" is not shown anymore, or have it checked by default and greyed out?

@maxwxyz, I wouldn't mark this as a bug by the way. I think this has been the case forever in FreeCAD. I do agree that it would be better if we have a proper solution for it.

krushia, 2024-07-24:
> ... the problem is that, for example, PropertyBoolList is _hidden by default_ in the property view, yet it has a status checkbox that you can mark as being hidden?

In part yes. What I'm confused about now (after @wwmayer's comment) is if this is because there is no such thing as "Status" for these property types. If that is true, then I believe there are **two bugs** here:

1. [BUG] Some property types don't support status bits
2. [BUG] A Status bit menu is shown for Property types that don't support status bits

> If so, what would be a proper solution to this? Change the status menu to make sure "[ ] Hidden" is not shown anymore, or have it checked by default and greyed out?

Some parameters have "read-only" status bits that are marked with an asterisk. This is what I would expect to see:
[image]