FreeCAD/FreeCAD #15033: render mass of object

csb2024, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I want batch draw 10,000 equal sized orbs (just in different positions) in FreeCAD view, and I find it's much slower, does anyone have a way to speed up the drawing? I am using topods_shape with compound type.

### Full version info

```shell
[code]
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * Rocket 3.3.0
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
please use the forum for help requests

closed 2024-06-29 by maxwxyz

csb2024, 2024-06-29:
repeat issue