FreeCAD/FreeCAD #15037: BIM: Hide all except selection not working

maxwxyz, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've tried BIM WB with the new example file and in the Views manager there is a button to hide everything except the selection.
This button just hides everything. Manually activating visibility for the building marks everything for recompute.
Not sure if I'm using it correctly but I expected that everything except the current level would be hidden.

Demo:
https://github.com/FreeCAD/FreeCAD/assets/6246609/f0f17b24-c342-4a66-8963-dc8510c4f7c5

BTW also creating a new level in the Views manager with the building selected creates it outside of the building and has to be manually dragged in the building in the tree view (not possible in the views manager)

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.26.1
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.20
  * Curves 0.6.40
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
@yorikvanhavre FYI

kaiwas, 2024-06-29:
> BTW also creating a new level in the Views manager with the building selected creates it outside

This annoying feature is present everywhere (globally). Each new element is created at the bottom of the list. If you have 1000 objects, you will have to scroll them all to move the 1001st to the right place.

yorikvanhavre, 2024-07-01:
I'm working on a big refactor of the BIMViews panel, and attacking these points too. But it will land after 1.0...

closed 2025-03-01 by maxwxyz