FreeCAD/FreeCAD #15038: BIM: Layers manager not working

maxwxyz, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The layers manager in BIM is not opened when clicking on the button:
[image]

It works normally in the Draft WB but in BIM WB nothing happens.

BIM WB calls `Gui.runCommand('BIM_Layers',0)` whereas Draft WB calls `Gui.runCommand('Draft_LayerManager',0)` not sure though if these are the same "layers" - but nothing happens in BIM regardless if there are layers in the document or not.
Tested with the new demo file.

@yorikvanhavre FYI

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.26.1
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.20
  * Curves 0.6.40
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-07-01:
I just tested with the same file and everything works normally for me... Any error message?

FYI the BIM layers tool is the same as the Draft one except that it adds a NativeIFC checkbox

maxwxyz, 2024-07-01:
No errors or messages, it just does not do anything.

furgo16, 2024-07-01:
Cannot reproduce with:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37943 (Git) Snap 1063
Build type: Release
Branch: main
Hash: 554310a6ba9217cf3aaf2b301772d8b4ce50db9f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

The layers manager window opens normally using BIMExample.FCStd. It's empty, as there are no layers in that model, as far as I can tell.

maxwxyz, 2024-07-01:
It works on my Windows machine as well, maybe it is a macOS problem?
@JohnOCFII could you test this and open the layers manager from BIM WB on macOS?

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.25.17
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.19

```

JohnOCFII, 2024-07-01:
I've never used BIM before --- but this is what I saw today when I clicked on the "Manage Layers" button:

```
Running the Python command 'BIM_Layers' failed:
Traceback (most recent call last):
  File "/Applications/Other Apps/FreeCAD-37928.app/Contents/Resources/Mod/BIM/bimcommands/BimLayers.py", line 90, in Activated
    self.dialog.buttonAssign.setIcon(QtGui.QIcon(":/icons/button_right.svg"))
    ^^^^^^^^^^^^^^^^^^^^^^^^

'PySide2.QtWidgets.QDialog' object has no attribute 'buttonAssign'
```

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.6.22
```

yorikvanhavre, 2024-07-02:
This is weird, but it might be due to a conflict with the Draft layers dialog... I'll make a PR for that

PR #15112 (2024-07-08): BIM: Fixed possible filename ambiguity in Layers
fixes #15038
closed 2024-07-08 by chennes