FreeCAD/FreeCAD #15040: Sketcher selection box barely visible in Light theme

MiqCG, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In revisions 37928 (and at least 37898) with Light theme Stylsheet, when selecting a sketcher geometry, the default colour of the selecting box is white and is very difficult to see it. 
In revision 37841, the selecting box was dark and therefore properly contrasted. 

I have searched in Github for [is:issue selection window sketcher light] with no relavant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/5e370ab5-c52f-470f-ae19-8f09cbe4d100

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
@MisterMakerNL FYI

MisterMakerNL, 2024-06-29:
It's black on my side, but I indeed run that selection color fix.
IT's also black in the dark theme, here it is visible but could definitely be improved.
I do not have a clue on how to change this color tho.

Syres916, 2024-06-29:
A new user picks the outer border of the rubber band selection from https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/GLPainter.cpp#L226-L229

Syres916, 2024-06-29:
For a quick fix before 1.0 release, a color that works across themes might be something like https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Sketcher_Rubber_Band_Border

After 1.0 release then a polished specific color mechanism for Sketcher Edit Mode rubberband could be created and made theme aware but IMHO it's too close to release to be trying that beforehand.

MisterMakerNL, 2024-06-30:
Couldn't we set this true the preference packs?

Syres916, 2024-06-30:
I assume you mean 'through' the preference packs, it can be done _either_ Stateful in stylesheets _or_ Parameter in the cfg files but more important would be whether the rubber band in Sketcher Edit Mode and rubber band in the main 3D View be configured the same or separately, at present they are separate.

luzpaz, 2024-07-25:
Is this ticket still relevant ?

maxwxyz, 2024-07-26:
Yes, the seletion box is white on a white background. You cannot see it.

MisterMakerNL, 2024-07-26:
> For a quick fix before 1.0 release, a color that works across themes might be something like [main...Syres916:FreeCAD:Sketcher_Rubber_Band_Border](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Sketcher_Rubber_Band_Border)
> 
> After 1.0 release then a polished specific color mechanism for Sketcher Edit Mode rubberband could be created and made theme aware but IMHO it's too close to release to be trying that beforehand.

> For a quick fix before 1.0 release, a color that works across themes might be something like [main...Syres916:FreeCAD:Sketcher_Rubber_Band_Border](https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Sketcher_Rubber_Band_Border)
> 
> After 1.0 release then a polished specific color mechanism for Sketcher Edit Mode rubberband could be created and made theme aware but IMHO it's too close to release to be trying that beforehand.

Could we go to blue? I tested the colors you used but it is green, and we have green lines in sketcher for fully defined lines.
I made it blue:
[image]
[image]
Used:
    rgb_r = 0.27f;
    rgb_g = 0.4f;
    rgb_b = 1.0f;
    rgb_a = 1.0f;
It's the last color from this list: https://dev.to/finnhvman/which-colors-look-good-on-black-and-white-2pe6

FEA-eng, 2024-08-03:
@maxwxyz This was fixed by #15586 so the issue can be closed.

closed 2024-08-03 by luzpaz