FreeCAD/FreeCAD #15041: Tool is not selected in the toolbar after clicking

nazar-pc, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When selecting tool in the toolbar, let's say "Create line", I can see from cursor that it is active, but it is not selected in the toolbar, which is unexpected and inconvenient, especially now that one icon actually has multiple modes all with slightly different icons used for cursor and do not necessarily match what is shown on the toolbar. This would be especially useful for newcomers.

I'm giving example of Sketcher, but I believe it affects virtually all workbenches.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.5.16
  * boltsfc 2022.11.5
  * fasteners 0.5.24
  * parts_library
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-29:
@kadet1090 is this possible, to create a toggle state as long as a tool is active? E.g. as long as the Pad dialog is open the pad icon appears pressed, same with line tools in sketcher.

kadet1090, 2024-06-30:
I don't think that it is trivial but certainly doable.

luzpaz, 2025-05-11:
Is this still relevant ?

nazar-pc, 2025-05-12:
Yes, in fact I tried latest weekly build and things got worse: the tool is no longer selected on the toolbar even if clicked on a toolbar. On the bright side there are tool parameters in the sketcher side bar, which does provide some indication of the tool used, but I'd expected it to be selected on the toolbar as well.

```
OS: Ubuntu 24.04.2 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/83.3641/1.25
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * Curves 0.6.42
```

luzpaz, 2025-05-12:
@maxwxyz this seems more like a bug than feature.

maxwxyz, 2025-05-13:
@luzpaz I disagree, as it never worked like this in FreeCAD. This is a proposed feature, that the icon of the current active tool appears with a different style on the taskbar.

maxwxyz, 2025-12-07:
IMO same as the snapping tools in Draft. Just create a PR. @FreeCAD/design-working-group any remarks?

pierreporte, 2025-12-07:
Maybe @obelisk79 has an idea since he’s working on a new icon theme?

kadet1090, 2026-01-05:
Everything we need is already present in code (look at draft and snapping) - we simply need to mark tools as active in code and that should be it.