FreeCAD/FreeCAD #15042: "Undo" doesn't work properly for external geometry creation

nazar-pc, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using "Undo" right after creation of external geometry doesn't actually "Undo" its creation, it only reverts some part of it partially and I have seen weird things happening after that where I had to close FreeCAD and redo things from scratch afterwards since it ended up in some weird incosistent state afterwards (sketch broke after a few steps completely).

I suspect it is somehow done in multiple steps internally, where first external edge is "created" and then "linked" to something, but still as a user I expected that operation to be atomically reverted when I use "Undo".

Haven't seen this reported yet, but seems important for this operation to work consistently.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.5.16
  * boltsfc 2022.11.5
  * fasteners 0.5.24
  * parts_library
```

### Subproject(s) affected?

Sketcher

### Anything else?

Recorded a screencast demonstrating the problem: 

https://github.com/FreeCAD/FreeCAD/assets/928965/84042397-b50c-4375-9fb3-e6c702779485

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-06-29:
A few more details, trying to delete external edge after such "Undo" results in this:
```
15:06:41  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to delete an external geometry with the given index: 0
15:06:41  Core#PcieSlotPanelMount: App.getDocument('Core').getObject('Sketch004').delExternal(0)
15:06:41  ViewProviderSketch: Not able to delete an external geometry with the given index: 0
```

Looks like UI is not synced with some internal data structures or something.

`File "<string>", line 1, in <module>` is also not particularly helpful for debugging purposes :upside_down_face:

maxwxyz, 2024-06-29:
is this also happening in the stable 0.21.2 version?
@PaddleStroke @FlachyJoe FYI

nazar-pc, 2024-06-29:
Honestly never used stable before and this project created with dev build fails to load in it :see_no_evil:

nazar-pc, 2024-06-29:
Wanted to test on stable, but couldn't due to https://github.com/FreeCAD/FreeCAD/issues/12978#issuecomment-2198225161

PaddleStroke, 2024-07-01:
I can reproduce.
I have just tested on 0.20 and the bug is already there.

My guess is that there is a problem in the commands opening/closing.
Probably the creation of external opens a command. Then the linking opens another command before the first is commited. So the first command is not on the undo stack. So when you undo it will undo the linking, then undo again will undo whatever you have done before.

closed 2024-07-12 by chennes