FreeCAD/FreeCAD #15045: TechDraw: No default page template

FEA-eng, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

A quite important regression (with respect to 0.21.2) regarding TechDraw is that there's no default drawing template and the page is initially empty. Many users will (and already do) find it undesirable. Also, this renders #13005 much less relevant. IMO there should be a default template like in the previous versions instead of a confusing empty page. That's also how it works in other CAD software.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-29:
While I agree that perhaps a blank A4 landscape page might not be to some/many users' taste, the default template is specified but not until the user clicks the 'Template' link:

[image]

and then it gives (on Linux) the template as _/home/username/freecadbuildlocation/share/Mod/TechDraw/Templates/Default_Template_A4_Landscape.svg_

I'm guessing it's nolonger hard coded in _Preferences_ due to the AppImage builds having a constantly changing folder structure.

FEA-eng, 2024-06-29:
@Syres916 On Windows, when I choose _Default_Template_A4_Landscape_, I get an empty page (unlike with other templates).

@WandererFan

maxwxyz, 2024-06-29:
This blank page template was added intentionally a few weeks ago. Which should be the default, I guess it depends on locale and used standard so this cannot be hardcoded? Maybe just rename the default one to 'A4 empty' to avoid confusion.

WandererFan, 2024-06-30:
There is a default template.  It is a blank page, like Ax_Landscape_blank.svg or ANSIx_Landscape_blank.svg, etc.

FEA-eng, 2024-07-01:
> There is a default template. It is a blank page

Yes, but I consider it a regression since in previous versions the default template had a drawing table and it was useful.

PaddleStroke, 2024-07-01:
I agree it's a regression.
imo the new page tool needs a taskbox instead of the 2 tools we now have, where you can select the size and standard of page. That requires quite some work as it needs to implement translation of template files and reorganisation of templates.

luzpaz, 2024-07-21:
> I agree it's a regression.

 Should I remove the 'Feature' tag in that case ?

maxwxyz, 2024-07-21:
IMO it's a feature as currently an empty page without border and title block is the default template. In the stable version its a ISO style template A4. I think depending on the preferences (ISO, ASME,...) a default page template should be set, for those who don't work with ISO or A4 size. Anyone can manually set a different template as default.

benj5378, 2024-09-18:
I also agree that it's a regression. I think the ISO should be default.

Just because some people work with ASME, it should be blank? So it works for neither ASME or ISO people? Nah

WandererFan, 2024-10-19:
Not a regression, just a change.  Nothing to do with ISO/ASME.

I use the blank one more that the ones with fields, but I suspect I am not a typical user.

The default template is "Default_Template_A4_Landscape.svg".  If anybody wants to do a PR that copies, say "A4_Landscape_TD.svg", to the default file spec, I have no objections.

There is also the preference for default template on the General tab.

closed 2024-10-30 by WandererFan

FEA-eng, 2024-11-04:
Shouldn't this stay open for now? The change below could be done to fix it:

> If anybody wants to do a PR that copies, say "A4_Landscape_TD.svg", to the default file spec, I have no objections.

maxwxyz, 2024-11-04:
IMO the blank one is ok as there is no default template depending on the used standard and projection method. Maybe the default template should sometimes be linked to the ISO/ASME preference