FreeCAD/FreeCAD #15048: task overlay interferes with spreadsheet

brettowe, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

while testing 0.22-dev I found that the new autohiding task overlay thing doesn't go away correctly at times
the biggest issue is if you load up a spreadsheet it will still 'exist' until you take your mouse off to the side for it to go away.
when its up the user will just feel that the spreadsheet isn't accepting clicks
at times it will be just hidden enough you don't realize that it is even there.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
Is this a duplicate to https://github.com/FreeCAD/FreeCAD/issues/12620 or https://github.com/FreeCAD/FreeCAD/issues/11637

luzpaz, 2024-07-02:
Ping @brettowe please respond if this a duplicate

brettowe, 2024-07-03:
sorry I didn't realize you wanted my input on this.
its close to #12620 but since I'm not a dev I can't say by how much.
feels partly like a timing issue like the UI isn't telling the overlay to go away soon enough when spreadsheet is there
a quick work around might be just to have it where spreadsheet is the active tab that overlay is turned off

luzpaz, 2024-07-07:
@maxwxyz maybe we should have an 'overlay' label ? (to keep track of these)

maxwxyz, 2024-07-07:
I'm fine with this. Or group these all in a project?

closed 2025-07-09 by brettowe