FreeCAD/FreeCAD #15049: No visual cue to help selecting a point in model view

theosib, 2024-06-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Have a look at the image here in the model view. I took a screen shot where the mouse was hovering over the line, and that line turns blue indicating that if I click, that line will be selected. Similar happens for faces.

But for points, there is no visual cue.

Say I want to select three points so that when I create a sketch, its plane will be defined by three points. This is nearly impossible to do since there's no way to tell when I'm lined up with the point, and I'm likely to select something else instead.

[image]

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
I cannot reproduce. Preselection and selection color is shown on my system

```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.26.1
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.20
  * Curves 0.6.40

```

theosib, 2024-06-30:
I'm not sure what to tell you. I don't see any pre-selection. It might be that the mouse cursor is getting in the way. Please tell me how to change the mouse cursor to something that doesn't get in the way, and I'll see what happens then. Thanks!

theosib, 2024-06-30:
[icoso.FCStd.zip](https://github.com/user-attachments/files/16044674/icoso.FCStd.zip)

I've experimented with making a sketch and selecting a point in the model view, and different themes have different results. Yellow on white and cyan on white are almost completely invisible. So maybe consider more contrast. But that's not what's happening for me.

Try selecting a point in the model as it is displayed, just using any theme. There is absolutely no pre-selection cue whatsoever.

maxwxyz, 2024-06-30:
what theme are you using when this happens and also what vertex size is set? Do you have a HiDPI monitor?

theosib, 2024-06-30:
I've tried multiple themes. and they all have this problem. If you have something simple, I can see the point pre-select. But if multiple points from multiple sketches are coincident, I think that's when the problem arises. Like one of multiple points is pre-selected, but the one lit up is not the one on top. The fix would be to highlight the front-most one from the viewer's pespective.

I'm not sure which setting you're referring to for the vertex size. I have a pick radius of 16. A marker size of 15. A vertex size of 9. A pixel size threshold of 15. 

I do have a high DPI display. I'm on a Mac.

maxwxyz, 2024-07-01:
Any chance to test this on a normal DPI monitor?
Is it visible if you increase the Vertex size?

theosib, 2024-07-02:
All I have are Macs, a Windows 10 machine with Hi DPI, and a headless Linux machine. Maybe I can try running it later on this Raspberry Pi 4 I just god. ;)

But as I say, if there are no coincident points, then it's not very good, but at least it changes color. My case seems to be one where there are multiple points in the same spot, and maybe FreeCAD is pre-selecting one, but there's another point in front of it.

theosib, 2024-07-02:
Try hovering over one of the vertex points in the attached file.
[icoso.FCStd.zip](https://github.com/user-attachments/files/16060639/icoso.FCStd.zip)

maxwxyz, 2024-07-02:
I cannot select these outer points at all on Windows and also see no preselection. I cannot select some lines too, I guess there is too much geometry on top of each other.
Related: https://github.com/FreeCAD/FreeCAD/issues/10308

closed 2025-05-21 by maxwxyz