FreeCAD/FreeCAD #15052: Importing SVG (produced by InkScape) fails with "ordinal not in range(128)"

theosib, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Create an empty document.
2. Select File > Import...
3. Select the attached SVG file.
4. Select to import as geometry

Expected outcome:
- The file gets imported and converted to geometry.

Actual outcome:
The following error:
```
22:13:20  (qt.qpa.fonts) Populating font family aliases took 42 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
22:13:37  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/Applications/FreeCAD 2.app/Contents/Resources/Mod/Draft/importSVG.py", line 1788, in insert
    parser.parse(pyopen(filename))
  File "/Applications/FreeCAD 2.app/Contents/Resources/lib/python3.11/xml/sax/expatreader.py", line 111, in parse
    xmlreader.IncrementalParser.parse(self, source)
  File "/Applications/FreeCAD 2.app/Contents/Resources/lib/python3.11/xml/sax/xmlreader.py", line 123, in parse
    buffer = file.read(self._bufsize)
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Applications/FreeCAD 2.app/Contents/Resources/lib/python3.11/encodings/ascii.py", line 26, in decode
    return codecs.ascii_decode(input, self.errors)[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'UnicodeDecodeError'>: 'ascii' codec can't decode byte 0xe1 in position 2817: ordinal not in range(128)
```

[image]

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-06-30:
I'm just making sure that I attached the SVG and not the original PDF, so here is the file again.
[image]

theosib, 2024-06-30:
I also wasn't sure InkScape had converted the text to a path, so I made sure of that. Import fails with the same error. Here's the file with the text as paths.
[image]

krushia, 2024-06-30:
What is the filename? Does it (or the directory it resides in) have any funky characters?

krushia, 2024-06-30:
Nevermind my prior comment. It is using your locale on import. First time I tried, it imported without error. Then I set LC_ALL=C and confirmed it fails to import.

GOOD:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
**Locale: English/United States (en_US)**

FAIL:
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
**Locale: C/Default (C)**

Jookia, 2024-06-30:
Related to #9726 ?

luzpaz, 2024-09-02:
@theosib @krushia  
Can  you try the workaround by @yorikvanhavre in  from https://github.com/FreeCAD/FreeCAD/issues/9726#issuecomment-1578677696 ?

krushia, 2024-09-05:
`f = pyopen(filename, mode='b')` works here, but this only "fixes" SAX. The string munging (for example in `getContents()`) will still trip up. Either use SAX completely, or ask SAX what the encoding tag is for use in the other places via pyopen(filename, encoding='whateverSaxSaid') and _**hope**_ the underlying system can handle that (dangerous).

luzpaz, 2024-10-26:
cc @yorikvanhavre :point_up_2:

luzpaz, 2025-04-12:
@aprospero any chance you can also take a look at this ?

aprospero, 2025-12-06:
Thank's, @luzpaz for 'let-me-copilot-that for-you'. :)

sure, one could now tackle every minor and major problem arising from using an unsuitable locale - at least it seems that's what AI would do.

Another idea: one could invest some effort and help people make the most out of the features of their favorite operating system.

No, seriously. It is certainly legitimate and honorable to deal with the arising problems individually in order to make the locale completely freely selectable - but I wouldn't be the right guy for that. I don't (want to) know enough about Python and its text conversion intricacies. 
And also I am not an OS X expert. I couldn't design or even test an OSX FreeCAD launch script that could set a user friendly locale.

Just my 2cents. :)