FreeCAD/FreeCAD #15053: Part Workbench: Create sketch from three points, origin unrelated to any of the points

theosib, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

You can use the attached FCStd file just for your convenience.

Steps:
1. Select the Part workbench
2. Select three points from sketches in the document, or one point and one line segment
3. Create a sketch and indicated that you want to create it from three points.

Expected result:
- The sketch will be created along the plane specified by the points, and the origin of that plane will related to those points in some meaningful way, such as being at their center, which would seem to make the most intuitive sense.

Actual result:
- The sketch is created along the plane specified by the points, BUT the origin is way off in space somewhere, having no sensible relationship to the points selected.

Note:
- This creates additional problems. For instance, if I draw in a sketch such as described above and extrude it, attaching a ShapeString to the surface of the extrusion will yield unintuitive results for a ShapeString anchored at the origin, because the origin of the surface is arbitrary.
 
[icoso.FCStd.zip](https://github.com/user-attachments/files/16042995/icoso.FCStd.zip)

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
How should FC know your desired rotation? You can edit it in the attachment editor. It's the same result if you'd add a plane in PartDesign.
It would be fixed if https://github.com/FreeCAD/FreeCAD/issues/11133 is implemented, that's why I close this.

closed 2024-06-30 by maxwxyz

theosib, 2024-06-30:
Thank you for your feedback. Just a few clarifications.

I can see how the orientation might be something FreeCAD can't figure out. But the ORIGIN should automatically be something more sensible. It's not, and I think it would be a great quality-of-life improvement to fix that.

Secondly, I cannot find any place to set the utf-8 locale. I've looked through lots of settings in preferences, and I've tried googling it. Can you tell me where this setting is?