FreeCAD/FreeCAD #15054: Suppressing geometry triggers TNP

Jookia, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've created a rectangle with two cylinders padded on top followed by a final square pad on the right. It looks like this:

[image]

Here's the FreeCAD file: [tower.zip](https://github.com/user-attachments/files/16043004/tower.zip)

Setting LeftCylinder to suppressed and recomputing works: RightCylinder's attachment moves from Pad001 (LeftCylinder) [Face5] to Face6, which is the same as Body's Face6.

However Un-suppressing causes it to stay with Face6! This is now the side of the base and causes this:

[image]

Link Daily seems unaffected by this bug.

### Full version info

```shell
OS: Arch Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37842 (Git)
Build type: Release
Branch: jookia_testing
Hash: 0f9dcefb3f8a99969117ff60167e6a8fdb26706d
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67

(Note: Branch base is 20f5e8f662230434db34fa4bf7f751809a9353ad)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
@bgbsww FYI

bgbsww, 2024-07-09:
Confirmed.  In LS3, on the unsuppress, there is an auto change element reference that fires when you unsuppress that reconnects things.

In main, the initial auto change on the suppress happens, but the second one on the unsuppress does not.  That might be because we have failed to remove something from the element map that we should have, so we don't run the missing reference code.

PR #15333 (2024-07-15): Toponaming: Missing suppress property and code
Fix #15054  Add the necessary property to hold the pre-suppression shape so that it can be restored when unsuppressed.  Correct the Feature code to use it.
closed 2024-07-15 by chennes