FreeCAD/FreeCAD #15055: After recent upgrade of Qt6 and python3.12 on Archlinux, Draft and Arch WB get errors.

Novite5, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After Python upgrading to 3.12.4-1 and Qt6 to 6.7.2-1 on Archlinux, FreeCAD 0.21.2 always reports these errors:
`FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

NOTICE: Addon "/.local/share/FreeCAD/Mod/CfdOF.backup1717505385.1731074" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/.local/share/FreeCAD/Mod/ModernUI" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/home/weber/.local/share/FreeCAD/Mod/ExplodedAssembly.backup1718009899.56283" disabled by presence of ADDON_DISABLED stopfile
Sheet Metal workbench loaded
FEMbyGEN Workbench loaded
NOTICE: Addon "/.local/share/FreeCAD/Mod/kicadStepUpMod.backup1718708000.5987735" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/.local/share/FreeCAD/Mod/CfdOF.backup1717505385.1731074" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/.local/share/FreeCAD/Mod/ModernUI" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/.local/share/FreeCAD/Mod/ExplodedAssembly.backup1718009899.56283" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "r/.local/share/FreeCAD/Mod/kicadStepUpMod.backup1718708000.5987735" disabled by presence of ADDON_DISABLED stopfile
WARNING: radv is not a conformant Vulkan implementation, testing use only.
Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:426: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:504: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Draft/draftguitools/gui_trackers.py:639: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]", buf)[0]
/usr/lib/freecad/Mod/Draft/importSVG.py:416: SyntaxWarning: invalid escape sequence '\.'
  _num = '([-+]?[0-9]*\.?[0-9]+([eE][-+]?[0-9]+)?)'
/usr/lib/freecad/Mod/Draft/importSVG.py:732: SyntaxWarning: invalid escape sequence '\d'
  inks_ver_pars = re.search("\d+\.\d+", inks_full_ver)
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:935: SyntaxWarning: invalid escape sequence '\s'
  _command = '\s*?' + _op + '\s*?' + _op2 + '\s*?'
/usr/lib/freecad/Mod/Draft/importSVG.py:938: SyntaxWarning: invalid escape sequence '\.'
  _num = '[-+]?[0-9]*\.?[0-9]+'
/usr/lib/freecad/Mod/Draft/importSVG.py:1620: SyntaxWarning: invalid escape sequence '\('
  _val = '\((.*?)\)'
/usr/lib/freecad/Mod/Draft/importSVG.py:1621: SyntaxWarning: invalid escape sequence '\s'
  _transf = _op + '\s*?' + _val
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1620, in attach
    image = Draft.loadTexture(Draft.svgpatterns()['simple'][1], 128)
                              ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 789, in svg_patterns
    loadSvgPatterns()
  File "/usr/lib/freecad/Mod/Draft/draftutils/utils.py", line 763, in load_svg_patterns
    p = importSVG.getContents(file, 'pattern')
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/freecad/Mod/Draft/importSVG.py", line 1709, in getContents
    f = pythonopen(filename)
        ^^^^^^^^^^
<class 'NameError'>: name 'pythonopen' is not defined
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
    ^^^^^^^^^
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
Traceback (most recent call last):
  File "/usr/lib/freecad/Mod/Arch/ArchWall.py", line 1714, in setDisplayMode
    self.fset.coordIndex.deleteValues(0)
    ^^^^^^^^^
<class 'AttributeError'>: '_ViewProviderWall' object has no attribute 'fset'
'PySide6.QtGui.QImage' object has no attribute 'byteCount'
load_texture: unable to load texture
'PySide6.QtGui.QImage' object has no attribute 'byteCount'
load_texture: unable to load texture
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "<string>", line 49, in <module>
  File "/usr/lib/freecad/Mod/Start/StartPage/StartPage.py", line 684, in postStart
    FreeCADGui.activateWorkbench(wb)
<class 'KeyError'>: "No such workbench 'BIMWorkbench'"
/usr/lib/freecad/Mod/Arch/ArchSite.py:95: SyntaxWarning: invalid escape sequence '\['
  buf = re.findall("point \[(.*?)\]",buf)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:751: SyntaxWarning: invalid escape sequence '\#'
  marker = re.findall("<line x1=.*?stroke=\"\#ffffff\".*?\/>",svg)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:767: SyntaxWarning: invalid escape sequence '\#'
  svg = re.sub("<line x1=.*?stroke=\"\#ffffff\".*?\/>","",svg,count=1)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:795: SyntaxWarning: invalid escape sequence '\?'
  svg = re.sub("<\?xml.*?>","",svg,flags=re.MULTILINE|re.DOTALL)
/usr/lib/freecad/Mod/Arch/ArchSectionPlane.py:797: SyntaxWarning: invalid escape sequence '\/'
  svg = re.sub("<\/svg>","",svg,flags=re.MULTILINE|re.DOTALL)
/usr/lib/freecad/Mod/Arch/ArchAxis.py:454: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]",buf)[0]
/usr/lib/freecad/Mod/Arch/ArchRebar.py:559: SyntaxWarning: invalid escape sequence '\['
  pts = re.findall("point \[(.*?)\]",comp.writeInventor().replace("\n",""))
Traceback (most recent call last):
  File "/home/weber/.local/share/FreeCAD/Mod/BIM/./BimViews.py", line 195, in update
    if vm.isVisible() and (vm.tree.state() != vm.tree.EditingState):
                                              ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'PySide6.QtWidgets.QTreeWidget' object has no attribute 'EditingState'
`

### Full version info

```shell
[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: Chinese/China (zh_CN) [ OS: German/Germany (de_DE) ]
Installed mods: 
  * fasteners 0.5.24
  * CfdOF.backup1717505385.1731074 1.21.7 (Disabled)
  * ModernUI (Disabled)
  * ExplodedAssembly.backup1718009899.56283 (Disabled)
  * 3D_Printing_Tools
  * BCFPlugin 1.0.0
  * 3DfindIT 1.2.0
  * A2plus 0.4.65
  * shiboken2
  * BIM 2021.12.0
  * sheetmetal 0.4.20
  * parts_library.backup1717343908.3237505
  * ExplodedAssembly
  * Fcmcua 1.0.0
  * Behave-Dark-Colors 0.1.1
  * FEMbyGEN 2.1.0
  * kicadStepUpMod.backup1718708000.5987735 11.1.8 (Disabled)
  * Render 2024.6.5
[/code]
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jookia, 2024-06-30:
There have been some Qt6 fixes in 0.22. Could you try the latest weekly?

Novite5, 2024-06-30:
@Jookia Thank you very much! The weekly build do not report errors anymore. But there is no Arch WB. Shall I use BIM instead?

Jookia, 2024-06-30:
On Sun, Jun 30, 2024 at 12:57:54AM -0700, Novite5 wrote:
> @Jookia Thank you very much! The weekly build do not report errors anymore. But there is no Arch WB. Shall I use BIM instead?

Yes, they have been merged. :)

Novite5, 2024-06-30:
@Jookia Thank you again!

closed 2024-06-30 by Novite5