FreeCAD/FreeCAD #15056: STEP Export going wrong

Anga23, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I export a design to a STEP, the file includes an additional object.

Original FreeCAD:
[image]

STEP File: 
[image]

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * fasteners 0.5.21
  * FeedsAndSpeeds 0.5.0
  * MakerWorkbench 1.0.1
  * sheetmetal 0.4.17
```

### Subproject(s) affected?

None

### Anything else?

[Winkel.zip](https://github.com/user-attachments/files/16043750/Winkel.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
how did you export this file, I cannot reproduce. The STEP export looks exactly the same.
Maybe you have exported hidden objects.

[test.zip](https://github.com/user-attachments/files/16043999/test.zip)

closed 2024-06-30 by maxwxyz

Anga23, 2024-06-30:
@maxwxyz Sorry for the inconvenience, now I try it again with following settings and it's fine: 

[image]

I was confused by the 2nd object witch I coldn't see it in the original FC file....
Thanks for Your time and support. Have a nice weekend!