FreeCAD/FreeCAD #15059: Addon manager causes temporary ban from github (ratelimit)

diggit, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I open `Addon manager`, it gets stuck and after that github is not reachable from my machine for several minutes. No response to ping. I am seeing terminal being flooded with
```
QObject::killTimer: Timers cannot be stopped from another thread
```
My suspicion is that `Addon manager` somehow causes some rate limit being breached and thus github bans my IP of several minutes.

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37923 (Git)
Build type: Release
Branch: main
Hash: 3b304e5b1a62f3119d7311aeb736dcb53cb4faad
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * SheetMetal 0.4.18
```

### Subproject(s) affected?

None

### Anything else?

network capture (filtered with `ip.src == 140.82.121.3 || ip.dst== 140.82.121.3`)  [fc_addon_manager_github.pcapng.gz](https://github.com/user-attachments/files/16043968/fc_addon_manager_github.pcapng.gz)

Running on up to date Archlinux, walynad, sway.
Sheet metal WB was manually cloned and was not obtained via Addon Manager due to this issue.

<details>
<summary>stdout log</summary>

```
freecad                                                                                                                                                                                                                                                                                                                                                                                          9.92h  Sun Jun 30 11:42:54 2024
FreeCAD 0.22.0, Libs: 0.22.0devR37923 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
Previous cache process was interrupted, restarting...
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
Git is disabled, skipping git macros
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
QObject::killTimer: Timers cannot be stopped from another thread
[... 390 lines cut ...]
  File "/usr/lib/freecad/Mod/AddonManager/AddonManager.py", line 403, in reject
    self.cache_package(repo)
  File "/usr/lib/freecad/Mod/AddonManager/AddonManager.py", line 543, in cache_package
    self.package_cache[repo.name] = repo.to_cache()
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^
TypeError: 'NoneType' object does not support item assignment
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
```

</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-30:
I cannot reproduce on Windows

kadet1090, 2024-06-30:
I can reproduce, banned two times today :sweat_smile:

diggit, 2024-06-30:
With git enabled (`Disable git` not ticked)
stdout log
```
FreeCAD 0.22.0, Libs: 0.22.0devR37923 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Sheet Metal workbench loaded
The `.FCStd#` backup format is deprecated as of v0.21 and may be removed in future versions.
To update, check the 'Preferences->General->Document->Use date and FCBak extension' option.
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
Previous cache process was interrupted, restarting...
The cached device pixel ratio value was stale on window update.  Please file a QTBUG which explains how to reproduce.
```
[fc_addon_manager_github_git_enabled.pcapng.gz](https://github.com/user-attachments/files/16044000/fc_addon_manager_github_git_enabled.pcapng.gz)

diggit, 2024-06-30:
Ok now I tried it via mobile data and at least GH does not ban me, but `Addon manager` gets stuck on same percentage (41%,  Downloaded icon for OSH Automated DOcumentation)

diggit, 2024-06-30:
My wired connection ISP is Vodafone CZ and they are using CGNAT with DSlite (default). So my public IPv4 is shared with others and that could explain rate limit issue (even though I did not notice it anywhere else than with FreeCAD).

@kadet1090 could you share some details about your connectivity?

chennes, 2024-06-30:
@diggit are you self-compiling? If so, the latest main should have fixed the "getting stuck at OSH" issue on Qt6 -- it may also fix the temp-ban problem, I'm not 100% sure which rate-limiter is being tripped. None of this has anything at all to do with the "use zip download" setting -- we're not using git (the app) to access this data, we're using GitHub (the website) to pull down one specific file.

kadet1090, 2024-06-30:
> @kadet1090 could you share some details about your connectivity?

Hard to say if I'm behind ISP NAT - my ISP is Orange PL. However I'm also using arch and get basically same results - after opening addon manager I can't access github anymore. I doub that this is API rate limiting - more like some firewall on some part of the network triggers and cuts of all traffic from my IP as I also cant access github via SSH. 

So either GitHub changed their firewalls and they are blocking traffic from IPs making rapid requests or there is another firewall that we have in common that is doing exactly that.

Nontheless - tried again and cannot reproduce now, maybe github were under DDoS attack and had enabled more strict firewall rules.

obelisk79, 2024-07-02:
This issue is still present, using an AUR build. 

```
OS: EndeavourOS (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git)
Build type: Release
Branch: main
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```

chennes, 2024-07-03:
Is that a clean rebuild? In particular, I want to ensure that the Addon Manager's Python files got copied correctly. Current MD5sum of NetworkManager.py should be 6898c9fe2ad4ea03135d375591435565.

obelisk79, 2024-07-03:
hmmm, looks like the checksum here is : 2a165bb8303af497941997d4c5c8ca3d I'll try and force it on my install.

chennes, 2024-07-03:
OK, thanks. Then the rate limit issue was not related to the redirect issue (which is the one I fixed last week). It's odd to me that the rate limiter would affect different platforms differently, but perhaps we are just skating on the edge of it, and some platforms are slightly faster than others. I'll try to code up something to detect it.

chennes, 2024-07-07:
This is really probably a deeper issue: it's not a great idea for every copy of FreeCAD to do this scrape every time. I was planning on doing the cache server-side post-1.0, but it appears that I am going to have to do it now.

chennes, 2024-07-19:
Although I'd still like to make this process more efficient, I believe the issue with hitting GitHub's rate limiter has been resolved. I'm closing this and essentially converting it to a feature request.

closed 2024-07-19 by chennes