FreeCAD/FreeCAD #15061: Locale defaults to "C" on Mac, rather than utf-8, triggers bugs

theosib, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've seen a few bug reports now where someone's locale is set wrong, resulting in bugs. My guess is that, like me, people are just going with the default. I don't think most people are going to be aware that they need to modify the locale in order to get FreeCAD to work properly, so I think perhaps it might be a good idea for FreeCAD to install with a more sensible setting.

References:
https://github.com/FreeCAD/FreeCAD/issues/9726
https://github.com/FreeCAD/FreeCAD/issues/15053

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-06-30:
Actually, I can't find any place to change the locale. I've tried googling and looking through the preferences, and I'm honestly just not finding it.

If the locale is causing so much trouble, besides making it default to something sensible, it might also be a good idea to make this option easier to find.

theosib, 2024-06-30:
I found that if I change the Number Format (something I would not have expected to relate to a utf-8 locale), to "Selected language," then it changes my locale to "English/United States (en_US)". But it doesn't add the utf-8, and there doesn't seem to be an option to add it.

krushia, 2024-07-02:
This is a FreeCAD meta-bug for sure, and affects all operating systems. You should never need to change your locale to get a program to work (and certainly not if you use the C/POSIX locale!).

There are numerous places in FreeCAD code where it is assumed that the locale uses UTF-8 string encoding, and other places where it assumes the locale is a subset of UTF-8. There should probably be a "best practices" guideline written in this area, and then a systematic review of the codebase.