FreeCAD/FreeCAD #15065: Problem : Expression editor doesn't recognize tab character for selection completion

hooperbloob, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While I'm typing in the expression editor, a list of suggested candidate references appear below the text box - all good. 
However, if enough characters are entered to match with a single remaining suggestion I should be able to hit the tab key to confirm it as the choice.  This is common behaviour on most operating systems:

https://www.howtogeek.com/195207/use-tab-completion-to-type-commands-faster-on-any-operating-system/

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.4.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-06-30:
That may be true for a terminal or text editor, but here in a form tab is used as a widget selector.

kriswilk, 2024-09-27:
> That may be true for a terminal or text editor, but here in a form tab is used as a widget selector.

I came upon this issue because of a related one (#16848). Although I generally support the typical use of TAB to switch the focus, I agree 100% with @hooperbloob that, in the _specific context_ of having the search results list visible, the TAB key should perform an autocomplete. I'd say that in that context, autocomplete is the _expected_ behaviour.

To address the possibility that the user really wants to switch focus, they could press ESC to hide the search results and use the TAB key normally.

Frankly, I would take @hooperbloob's suggestion a step further: TAB should autocomplete the _first entry_ in the search list, not just if there is only one entry. Sometimes the correct entry is already at the top even if others are below it. I should be able to just type TAB and have it completed.

luzpaz, 2024-10-16:
I've pushed for this in the past. I think it's more intuitive to use tab. There are some discussions on the forum about it.