FreeCAD/FreeCAD #15066: Can't link a group in another document

nazar-pc, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a group with a few sketches, trying to link the whole group in another document, but getting this:
```
21:01:05  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: PropertyLinkList does not support external object
```

Looks like it was already mentioned on the forum (without any resolution) in https://forum.freecad.org/viewtopic.php?t=40706 years ago

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.6.22
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-06-30:
How are you linking? You can't put an external object directly into a Group (or even a LinkGroup) because they use `App::PropertyLinkList` to manage children. Only `App::PropertyXLinkList` supports such a thing.

You should be able to use a Link object to point to an external object, as that uses `App::PropertyXLink`.

nazar-pc, 2024-06-30:
I do not know what LinkGroup is yet, but I was trying to link the whole group element by selecting it and creating link in another document like I would with any other element like Sketch.

I can totall re-create the same set of groups and link individual elements from one document into another and then move under the group with identical name (which is annoying BTW, would be more convenient to just move it directly), it is just very tedious and more burdensome to maintain.

I am not familiar with codebase to understand difference between various kinds of links internally.

krushia, 2024-06-30:
Okay I see the bug, will. put example here in a few mins.

krushia, 2024-06-30:
Upon loading parent.FCStd again, the WorkaroundLink is shown as broken, but the BugLink shows as working.

maxwxyz, 2024-07-01:
So is this still a relevant issue?