FreeCAD/FreeCAD #15068: Problem : Sketcher - Delete key in non-functional in macOS

hooperbloob, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is likely the same problem as [15023](https://github.com/FreeCAD/FreeCAD/issues/15023). You can delete selected items in the Sketcher via the context menu but not via the delete key.

There are probably other editors in FreeCAD that have the same issue in macOS.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.4.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-01:
@wwmayer this seems to be the same but https://github.com/FreeCAD/FreeCAD/pull/15057 just fixes it in the Spreadsheet WB

wwmayer, 2024-07-01:
There are more things to check where Delete probably doesn't work:

1. Open the Parameter editor. Select a group item in the left window and press Delete. Select a value item in the right window and press Delete.

2. Open FemCalculixCantilever2D.FCStd of the examples directory. Double-click the ConstraintForce item in the tree view.
Select Plane:Edge3 and press the Delete key

3. Create a sketch and start to draw a rectangle. After clicking the first point you should get two spin boxes where one has the focus. Does pressing the Delete key clear the length value inside the spin box?

4. Material editor:
* Open the Materials panel and click on Launch editor.
* Select an existing material and click on Inherit New.
* Click on New Folder
* Select the New Folder item and press Delete
* Does a confirmation dialog appear?

5. Tech draw view
* Create a simple shape, e.g. a Part box
* Go to Tech draw and create a view
* Add the box to the tech draw view
* Select the item and press Delete
* Will it be deleted?

See also:
https://github.com/FreeCAD/FreeCAD/pull/15083
https://github.com/FreeCAD/FreeCAD/pull/15085

luzpaz, 2024-07-17:
See also #9639

luzpaz, 2024-08-18:
@hooperbloob check out [dirkolbrich](https://github.com/dirkolbrich) #15895  
CC @nino

luzpaz, 2024-09-24:
@hyarion can you review this open ticket in lieu of #16682 ? TIA

hyarion, 2024-09-24:
I've tested removing items from the 3d view and both constraints and elements from the task view in sketcher. All these works with Std_Delete mapped to backspace on macos.

I haven't tested the once that wwmayer listed in a comment (I got stuck on fem, as my setup doesn't work with fem currently)

I think we can close this ticket.

luzpaz, 2024-09-24:
@hyarion great. Is there a WIP effort for FEM and other remaining loose-ends ?

closed 2024-09-24 by luzpaz

hyarion, 2024-09-24:
@luzpaz maybe this issue isn't the correct place for discussing another PR?

But to answer your question:
The gist of the discussion during the PR and issue meeting was that it looked like a good fix and should be safe. It got merged to main and marked to be backported to 1.0 if no one finds any related issues using the weekly builds.

luzpaz, 2024-09-25:
@hyarion indeed, not ideal to discuss in closed issue (although are we discussing another PR?)  
My intention was/is to ascertain if there were/are loose ends that needed to be tracked in a new ticket (i.e. FEM, and fiends).

hyarion, 2024-09-25:
@luzpaz sorry, I missed the place op wrote "There are probably other editors...", I thought it was only a sketcher issue. I stand corrected :)

The loose ends is that the non-ticked checkboxes in this PR: https://github.com/FreeCAD/FreeCAD/pull/16682 has not been verified that everything works as expected.

The current state is that it is assumed that it will work, but it hasn't been tested.

I see two ways forward.
1. We continue to assume that it works and if someone proves us wrong with an issue, we'll fix it then
2. We test it, though I failed to find the places, so I need help in that case.

As I understood the meeting, the conclusion was the first path. Assume and if it fails fix then.

If you feel strongly that 2 is a better way forward, then feel free to test it or find some one who can find the places :)

luzpaz, 2024-09-25:
Thanks for the elaborate reply, @hyarion. I appreciate all the effort and patience. I wasn't present for the meeting so I'm out of the loop. Thanks for informing me.

For any macOS testers, here are the remaining workbenches that need to be tested  
https://github.com/FreeCAD/FreeCAD/pull/16682#issuecomment-2374395879