FreeCAD/FreeCAD #15069: Model tree : Delete key non-functional in MacOS

hooperbloob, 2024-06-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is likely the same problem as https://github.com/FreeCAD/FreeCAD/issues/15023 and https://github.com/FreeCAD/FreeCAD/issues/15068. You can delete selected items in the model tree via the context menu but not via the delete key.

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * sheetmetal 0.4.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-06-30:
The delete key does work but you need to hold down **fn** (function).

That is, the action in the code appears to be bound to delete (rightwards), not backspace-delete (leftwards).

This is a Mac/windows difference, at least. 

On a Windows machine, the key marked **delete** (as opposed to the backspace key) deletes the character to the right of the cursor. 

On a Mac, the key marked delete is a backspace-delete -- it deletes the character to the left of the cursor. 

Most (all?) Mac keyboards do not have a standalone delete key AFAIK.

From memory, I don't think this behaviour is a regression; I think it worked this way on 0.20 and 0.21.

wwmayer, 2024-07-01:
The tree view doesn't implement the deletion in its own key event handler but uses the global shortcut that triggers the command Std_Delete. This command has defined the shortcut `QKeySequence::Delete` that under [macOS](https://doc.qt.io/qt-6/qkeysequence.html#standard-shortcuts) is bound to Forward Delete or Meta + D.

So, there is nothing that can be done from our side. If you prefer the backspace-delete then you must override the shortcut of the Std_Delete command in the Customize dialog.

closed 2024-07-01 by maxwxyz