FreeCAD/FreeCAD #15082: AddonManager: empty widget warning for addonmanager_firstrun.py

Roy-043, 2024-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the 'firstrun' dialog is shown there is an empty widget warning:

`"Empty widget item in QVBoxLayout 'verticalLayout'."`

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-01:
@Roy-043 see https://github.com/FreeCAD/FreeCAD/issues/14415#issuecomment-2142790594
As far as I'm aware all horizontal lines cause this output in certain Qt versions, I've not built myself a Qt6.7.1 based FC yet to see if it's been dealt with.

chennes, 2024-07-19:
For reference, the warning does still occur in Qt 6.7.2.

PR #15511 (2024-07-22): Addon Manager: Silence warning due to use of horizontal line
Fixes #15082
closed 2024-07-22 by chennes