FreeCAD/FreeCAD #15095: [TNP] TNP names still visible on app::PropertyLink

shaise, 2024-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It looks like even after fix #14561 TNP names still appear in python workbenches. (App:PropertyLink still shows all the TNP name instead of just the base name)
[image]

see video how to reproduce:
https://github.com/FreeCAD/FreeCAD/pull/14561#issuecomment-2156134416

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37735 +262 (Git)
Build type: Release
Branch: CAM-Simulator3
Hash: ce225649ac8c712d3b34e95d7d2d59bedd648da1
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM_Workbench 2021.12.0
  * fasteners 0.5.24
  * Manipulator 1.5.7
  * OpenTheme 2024.6.22
  * PieMenu 1.5.1
  * Render 2024.1.23
  * SelectorToolbar
  * sheetmetal 0.4.21
  * TabBar (Disabled)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-07-01:
See #13248  This may not be "wrong", although there may be an inconsistency.  The key would be checking LS3 to see if the visibility occurs there as well.  If it does, then this is essentially a feature request ( and a good one IMHO ), but not a bug.

Syres916, 2024-07-02:
@shaise it's late and maybe I'm not thinking properly, as CAM is a Python workbench (with some C++ code) I would expect serious misbehaviour with changing a model, for instance adding an arc part way down a straight edge therefore gaining two extra edges.

Are you seeing the same TNP name info being passed in CAM when testing the new simulator or is CAM using a different method/property type from Sheetmetal, I'm really confused???

shaise, 2024-07-02:
Hi @bgbsww , Thank you again for your great work on TNP!

> See #13248 This may not be "wrong", although there may be an inconsistency.  If it does, then this is essentially a feature request ( and a good one IMHO ), but not a bug.

I totally agree that this should have been a good thing, and initially I was happy that I get these new names. The problem is that these names are not usable - or at least I don't know how to. I was under the impression that if an App::PropertyLink returns these face/edge names then I would expect that `Object.getElement(elementName)` will return the correct face/edge
However I get the following error:
`<class 'ValueError'>: Invalid subelement name` 
This function only works with the short name. (Perhaps its only a mater of updating this function?)
Anyway, regardless of all that it is now inconsistent in the sense that   App::PropertyLink sometimes returns short name and sometimes TNP names. I think it happens usually when  App::PropertyLink holds multiple elements. 

> The key would be checking LS3 to see if the visibility occurs there as well.

Here is how it looks on LS3 (and I guess that's why SheetMetal always worked on LS3): 
My workaround translation function from TNP to short is never used in LS3.
[image]

bgbsww, 2024-07-02:
Thank you for all this legwork, that helps a lot!

bgbsww, 2024-07-02:
Okay, so tracing back what happens with PropertyLinkSubs you quickly come to getSubNameWithStyle which is called by getSubValues.  Both of those have a boolean parameter for (drum roll) whether to return an old or new style name.

I think this is a call to getSubValues() somewhere that either has no parameter or the wrong value.  That'll be a visible difference between main and LS3 source code.

shaise, 2024-07-02:
Hi @bgbsww ,
Thank you for you answer!
What about `shape.getElement(elementName)`? If it can accept TNP names like LS3 then we can use them in Python as well - as you recommended.

bgbsww, 2024-07-02:
My bad, I missed that point:  That looks like ComplexGeoDataPy::getElementName or GeoFeature::getElementName  or SubObject::getElementName

The first two look complete and have been tested.  That last one also looks okay, but in src/App/DocumentObserver.cpp there is definitely missing some TNP element code that exists in LS3, and is the prime suspect.

This is going to need a python test for the getElementName that checks both the short and long names, and then the code moved in to support it.

All of the above is basically a note to future me, unless somebody steps in and patches this.

shaise, 2024-07-02:
Hi @bgbsww , Thanks.
I will take a look on the functions you mentioned and see if I can do something with it, and make tests as well. This is a huge project, I know, so anything I can do to help.

shaise, 2024-07-02:
Hi @bgbsww ,
Here are my findings regarding the shape.getElement() function.
The Py implementation is: `PyObject* TopoShapePy::getElement(PyObject *args)` in file TopoShapePyImp.cpp

the original code is:
```
    char* input;
    if (!PyArg_ParseTuple(args, "s", &input))
        return nullptr;

    boost::regex ex("^(Face|Edge|Vertex)[1-9][0-9]*$");

    try {
        if (boost::regex_match(input, ex)) {
            std::unique_ptr<Part::ShapeSegment> s(static_cast<Part::ShapeSegment*>
                (getTopoShapePtr()->getSubElementByName(input)));
            TopoDS_Shape shape = s->Shape;
            switch (shape.ShapeType()) {
                case TopAbs_FACE:
                    return new TopoShapeFacePy(new TopoShape(shape));
                case TopAbs_EDGE:
                    return new TopoShapeEdgePy(new TopoShape(shape));
                case TopAbs_VERTEX:
                    return new TopoShapeVertexPy(new TopoShape(shape));
                default:
                    break;
            }
        }

        PyErr_SetString(PyExc_ValueError, "Invalid subelement name");
        return nullptr;
    }
    catch (Standard_Failure& e) {
        PyErr_SetString(PartExceptionOCCError, e.GetMessageString());
        return nullptr;
    }*/
```

If I change it to realthunder's (rather simpler) code:
```
    char* input;
    PyObject* silent = Py_False;
    if (!PyArg_ParseTuple(args, "s|O", &input, &silent)) {
        return nullptr;
    }
    try {
        PyObject* res = getTopoShapePtr()->getPySubShape(input, PyObject_IsTrue(silent));
        if (!res) {
            Py_Return;
        }
        return res;
    }
    PY_CATCH_OCC
```

If I do so, this function now supports both short and TNP names with no issues. But I never touched freecad's Core code, so I'm a bit afraid to place a PR on it.
What do you think?

shai

bgbsww, 2024-07-02:
That's even better.  The note I made about missing code in src/App/DocumentObserver.cpp still stands, and I've added it to my checklist, but the fix you've found looks correct for this problem.

getPySubShape calls getSubTopoShape which is the correct element aware code.

I'd progress a PR, especially if you have a python level test ( or can make a quick one ) that tries getElement against both a long and a short name.

For the PR's I submit, I generally run the full c++ and python test suites on them, but the CI is supposed to make that redundant.

Nice work!

shaise, 2024-07-03:
Hi @bgbsww ,
Can you point me to a place where I can learn how to add tests? Or at least a sample of one. I'm new to it.
Thanks!

bgbsww, 2024-07-03:
Sure!  Since this is TNP related, it would be reasonable to add tests like this one to this file: [TopoTests](https://github.com/FreeCAD/FreeCAD/blob/04cc890fd0685bc4143de1a8000e92dce0580d28/src/Mod/PartDesign/PartDesignTests/TestTopologicalNamingProblem.py#L133-L158)

Running this test can be done from the Test workbench as `TestPartDesignApp`, or from the command line as `freecad -t TestPartDesignApp.TestTopologicalNamingProblem` or a single test as `freecad -t TestPartDesignApp.TestTopologicalNamingProblem.testPartDesignElementMapPad`

The `# arrange  # act  #assert`  pattern is a desirable standard in the testing world, and more details of available assertions are at https://docs.python.org/3/library/unittest.html

And I wrote and use this script to take a console log from an interactive session as a starting point for a test:
```
#! /bin/env python3

""" Take a FreeCAD console output and convert it into a test """

# Todo: convert this to a freecad macro and dialog
# Todo: figure out a way to get the console output in a python script.  Might require a some c++ to allow
#       the python to have an observer or the equivalent

import sys
import re

# Todo: Move these out of global constants into selectable values
KillGui = True
KillNewDoc = True
KillViewObject = True
NoEmptyLines = True
Indent = "    "
KillImport = True
KillActiveSketch = True

def main():
    infile = sys.argv[1]
    outfile = sys.argv[2]
    linenum = 0
    # Todo: Add a full prefix facility to insert some lines.
    lines=[ "def testSomething(self):\n", "    doc = App.ActiveDocument\n"]
    with (open(infile) as f):
        for line in f:
            linenum = linenum + 1
            if linenum < 3: # Todo: Don't assume a full log with garbage lines up front
                continue    # Get rid of first two lines with python startup
            if line[0:4] == ">>> ": # Get rid of all the interpreter prompts
                line = line[4:]
            if line[0:2] == "##":  # Get rid of all the comment lines
                continue
            if line[0:9] == "# Object ":  # Should have double ##, doesn't
                continue
            if line[0:2] == "# ":   # Make the actual code lines real
                line = line[2:]
            if KillGui and line[0:4] == "Gui.":
                continue
            if KillNewDoc and line[0:17] == "App.newDocument()":
                continue
            if KillNewDoc and line[0:21] == "App.setActiveDocument":
                continue
            #if KillNewDoc and line[4:18] == "ActiveDocument":
            #    continue
            if KillImport and line[0:7] == "import ":
                continue
            if KillActiveSketch and line.find("ActiveSketch") != -1:
                continue
            if KillViewObject and (
                    line.find("tv =") != -1 or
                    line.find("ViewObject") != -1 or
                    line.find(" tv.") != -1 or
                    line.find("tv.") == 0 or
                    line.find("(tv)") != -1 or
                    line.find(" tv:") != -1):
                continue
            line = re.sub("App.getDocument\('[^']*'\)","doc", line)
            line = re.sub("App.ActiveDocument","doc", line)
            line = re.sub(".getObject\('([^']*)'\)",".\g<1>", line)
            if NoEmptyLines and len(line.strip()) != 0:
                lines.append(Indent+line)
    # Todo: Add postfix to the body lines by asserting some things
    # Todo: Potentially write a dialog to ask which objects to assert against and give a menu of assertions.
    with open(outfile, "w") as f:
        f.truncate()
        for line in lines:
            f.writelines(line)

if __name__ == "__main__":
    main()
```

shaise, 2024-07-04:
@bgbsww ,
Thank you for your help! I have placed a pr #15177. I have added a test to TestTopologicalNamingProblem. Name:  testGetElementFunctionality.

bgbsww, 2024-07-04:
@shaise Thanks a bunch for picking up and running with this.  The extra hands are very much appreciated.

closed 2024-07-14 by chennes

shaise, 2024-07-14:
so, are we ok with TNP names shown in the task panel? this might confuse the users.

commit 2e39b416 (2024-07-14): Merge pull request #15177 from shaise/Fix-Python-getElement

[TNP] Fix python get element. partially fixes #15095
commit b0871411 (2024-07-14): Fix python shape setElement function. Partially fixes #15095