FreeCAD/FreeCAD #15100: Assembly Gear linkage failure: 'PySide2.QtWidgets.QWidget' object has no attribute 'limitCheckbox1'

johnsonm, 2024-07-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using the CrashGear.zip in https://github.com/FreeCAD/FreeCAD/issues/14458#issuecomment-2182669826 when I open SingleAssembly.FCStd in current `main` with submodules updated, activate the assembly, select the left circle then the right circle (either by radius or by top face) and the Gear linkage, I get:

```
Running the Python command 'Assembly_CreateJointGears' failed:
Traceback (most recent call last):
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/CommandCreateJoint.py", line 413, in Activated
    activateJoint(11)
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/CommandCreateJoint.py", line 61, in activateJoint
    panel = TaskAssemblyCreateJoint(index)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/home/johnsonm/d/FreeCAD/build/Mod/Assembly/JointObject.py", line 1301, in __init__
    self.form.limitCheckbox1.stateChanged.connect(self.adaptUi)
    ^^^^^^^^^^^^^^^^^^^^^^^^

'PySide2.QtWidgets.QWidget' object has no attribute 'limitCheckbox1'
```

[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Unknown
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.5.24
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-07-02:
Re-confirmed with a rebuild at 04cc890fd0685b

PaddleStroke, 2024-07-03:
This is a build issue. I don't know why but whenever the .ui files are updated (here the TaskAssemblyCreateJoint.ui file) then the qrc file needs to be manually deleted or the build will not rebuild the widget.

So you have to go to your build folder : 
FC build\src\Mod\Assembly\Gui
and delete Assembly_rc.py and qrc_Assembly.cpp
Then build again.

Alternatively you can clean the project and rebuild.

PaddleStroke, 2024-07-03:
@maxwxyz this can be closed.

closed 2024-07-03 by maxwxyz

johnsonm, 2024-07-03:
OK, sounds like actually a CMakefile bug of not expressing a dependency. But I figured other .ui file updates had probably happened since I last did a clean build, so I should probably delete all those files, not just those two, before rebuilding, so that I don't run into other similar problems.

```
find . -name \*_rc\* -o -name \*qrc_\* -delete -print
```

That deleted lots of files, including the two you mentioned.

After that I re-ran cmake and make, and I still see the problem. So it looks like there's still something missing and I'll have to do a clean build.

PaddleStroke, 2024-07-03:
I'm not good at cmake so I don't know how to fix this. But if you know how to fix it I'd appreciate it very much. This has been annoying me a lot.

johnsonm, 2024-07-04:
I'm afraid that cmake isn't my specialty either. And given that deleting those files (as well as others) didn't fix the problem, I don't have any idea what to look for. But just confirming that the clean build resolved the problem.