FreeCAD/FreeCAD #15102: Reliable crash when changing chamfer size

theosib, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm not sure how to describe the issue other than to provide a sample file and steps to reproduce.

1. Load the attached FCStd file.
2. Double click what I have named "glitchy chamfer".
3. Edit the size and set it to 3.9.

FreeCAD crashes every time.

[try3.FCStd.zip](https://github.com/user-attachments/files/16060494/try3.FCStd.zip)

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-02:
The file itself seems to be corrupted, there are some errors when opening.
[image]

Syres916, 2024-07-02:
@theosib could you please use the latest weekly build now at 38001, every time I open any of submitted files my machine is brought to a grinding halt.

The Seg fault traceback on my box is:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f384c3b3090]
#1  0x7f384e717316 in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0xe6
#2  0x7f384e717ef7 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0x297
#3  0x7f38238d07d1 in Sketcher::SketchObject::buildShape() from /home/john/freecad-main-build763/Mod/Sketcher/Sketcher.so+0x231
#4  0x7f38238d1514 in Sketcher::SketchObject::execute() from /home/john/freecad-main-build763/Mod/Sketcher/Sketcher.so+0xd4
#5  0x7f384e4fb3ea in App::DocumentObject::recompute() from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0xba
#6  0x7f383657d78d in Part::Feature::recompute() from /home/john/freecad-main-build763/Mod/Part/Part.so+0xd
#7  0x7f384e49208d in App::Document::_recomputeFeature(App::DocumentObject*) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0x33d
#8  0x7f384e498f20 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0x630
#9  0x7f384e56556a in App::DocumentPy::recompute(_object*) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0x2ca
#10  0x7f384e56576c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /home/john/freecad-main-build763/lib/libFreeCADApp.so+0x2c
#11  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x2a8697) [0x7f384dc6e697]
#12  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyObject_MakeTpCall+0xab) [0x7f384dc6eb2b]
#13  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x74df3) [0x7f384da3adf3]
#14  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalFrameDefault+0x7d86) [0x7f384da42ef6]
#15  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(_PyEval_EvalCodeWithName+0x8fb) [0x7f384db90e4b]
#16  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCodeEx+0x42) [0x7f384db911d2]
#17  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyEval_EvalCode+0x1f) [0x7f384db915bf]
#18  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(+0x18c05c) [0x7f384db5205c]
#19  /lib/x86_64-linux-gnu/libpython3.8.so.1.0(PyRun_StringFlags+0x79) [0x7f384db523a9]
#20  0x7f384e0ad19c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/john/freecad-main-build763/lib/libFreeCADBase.so+0x6c
#21  0x7f384ef6687c in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x8c
#22  0x7f384ef66a23 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x103
#23  0x7f384ef743fd in StdCmdRefresh::activated(int) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x6d
#24  0x7f384ef6d29e in Gui::Command::_invoke(int, bool) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x3de
#25  0x7f384ef6d56f in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x11f
#26  0x7f384caa7328 in QMetaObject::activate(QObject*, int, int, void**) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x928
#27  0x7f384d48d3e6 in QAction::triggered(bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#28  0x7f384d48faa2 in QAction::activate(QAction::ActionEvent) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf2
#29  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x260e10) [0x7f384d589e10]
#30  0x7f384d58a035 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe5
#31  0x7f384d67d5fe in QToolButton::mouseReleaseEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#32  0x7f384d4d62b6 in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x286
#33  0x7f384d67d6a8 in QToolButton::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x38
#34  0x7f384d493a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#35  0x7f384d49d343 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x583
#36  0x7f384ef1c2c8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/john/freecad-main-build763/lib/libFreeCADGui.so+0x98
#37  0x7f384ca7b80a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x18a
#38  0x7f384d49c457 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#39  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c935d) [0x7f384d4f235d]
#40  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cc1ec) [0x7f384d4f51ec]
#41  0x7f384d493a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x86
#42  0x7f384d49d0f0 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x330
[... cut ...]
I'll try and edit file outside of FreeCAD and resubmit a useable file.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1
```

theosib, 2024-07-02:
I opened in 380001, and this is what printed to the console panel:
```
08:10:54  (qt.qpa.fonts) Populating font family aliases took 43 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Body
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch002
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch003
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket002
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket002
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Fillet
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Fillet
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch004
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket003
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket003
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Fillet001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Fillet001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch005
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Body001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch006
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad002
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad002
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch007
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket004
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket004
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch008
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket005
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket005
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch009
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket006
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pocket006
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch010
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad003
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad003
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Mirrored
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Sketch011
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad004
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad004
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Binder
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad005
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad005
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Binder001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad006
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Pad006
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Mirrored001
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Chamfer
08:11:05  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: try3#Chamfer
08:11:05  Reading failed from embedded file: StringHasher.Table.txt (1132 bytes, 304 bytes compressed)
08:11:06  SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
```

theosib, 2024-07-02:
Also the new release has broken my ability to use my design. It comes up and indicates that everything need to be recomputed, so I tell it to recompute, and it immediately crashes. I think I have to revert to the prior release.

theosib, 2024-07-02:
I ran freecad with lldb. It looks like it has a memory corruption issue.
```
<PropertyLinks> PropertyLinks.cpp(435): try3#Sketch001.AttachmentSupport auto change element reference try3#Pad ;g85139;SKT;:He64,E;:G;XTR;:He64:7,F.Face3 -> ;g27;SKT;:He64,E;:G;XTR;:He64:7,F.Face3
<PropertyLinks> PropertyLinks.cpp(435): try3#Sketch001.ExternalGeometry auto change element reference try3#Pad ;g85234;SKT;:He64,E.Edge15 -> ;g29;SKT;:He64,E.Edge15
<PropertyLinks> PropertyLinks.cpp(435): try3#Sketch001.ExternalGeometry auto change element reference try3#Pad ;g85139v2;SKT;:He64,V;:G;XTR;:He64:7,E.Edge8 -> ;g27v2;SKT;:He64,V;:G;XTR;:He64:7,E.Edge8
<PropertyLinks> PropertyLinks.cpp(435): try3#Sketch001.ExternalGeometry auto change element reference try3#Pad ;g85139;SKT;:He64,E;:G;XTR;:He64:7,F;:U;XTR;:He64:7,E.Edge10 -> ;g27;SKT;:He64,E;:G;XTR;:He64:7,F;:U;XTR;:He64:7,E.Edge10
Process 32232 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x30)
    frame #0: 0x0000000102d31c98 libFreeCADApp.dylib`std::__1::__tree_node_base<void*>*& std::__1::__tree<std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::__map_value_compare<Data::MappedName, std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<Data::MappedName, Data::IndexedName> > >::__find_equal<Data::MappedName>(std::__1::__tree_end_node<std::__1::__tree_node_base<void*>*>*&, Data::MappedName const&) + 8
libFreeCADApp.dylib`std::__1::__tree<std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::__map_value_compare<Data::MappedName, std::__1::__value_type<Data::MappedName, Data::IndexedName>, std::__1::less<void>, true>, std::__1::allocator<std::__1::__value_type<Data::MappedName, Data::IndexedName> > >::__find_equal<Data::MappedName>:
->  0x102d31c98 <+8>:  ldr    x16, [x0, #0x8]!
    0x102d31c9c <+12>: cbz    x16, 0x102d31e74          ; <+484>
    0x102d31ca0 <+16>: ldp    x8, x9, [x2]
    0x102d31ca4 <+20>: ldrsw  x10, [x8, #0x4]
```

theosib, 2024-07-02:
Also, it opens fine in the previous release. I don't think my design process was especially remarkable. I just built some things in Part Design. I didn't even do much retro modification earlier in the tree, and it's hard to see how a normal design process should result in a corrupted file. So either (a) the previous release produced a bad file that it somehow can manipulate or (b) there's a regression in 38001.

theosib, 2024-07-02:
BTW, 38001 shows this on the console before I even load a file:
(qt.qpa.fonts) Populating font family aliases took 45 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost.

theosib, 2024-07-02:
Finally, I attached lldb to 37898 release and reproduced the original crash. Here's what I get from lldb:
```
lldb freecad
(lldb) target create "freecad"
Current executable set to '/Applications/FreeCAD 2.app/Contents/Resources/bin/freecad' (arm64).
(lldb) run
Process 32481 launched: '/Applications/FreeCAD 2.app/Contents/Resources/bin/freecad' (arm64)
FreeCAD 0.22.0, Libs: 0.22.0devR37898 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.qpa.fonts) Populating font family aliases took 42 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
2024-07-02 08:50:05.007152-0400 freecad[32481:52521413] [default] CGSWindowShmemCreateWithPort failed on port 0
Reading failed from embedded file: StringHasher.Table.txt (1132 bytes, 304 bytes compressed)
2024-07-02 08:50:10.006754-0400 freecad[32481:52521413] [default] CGSWindowShmemCreateWithPort failed on port 0
SketchObject::onChanged(): Unmanaged change of Constraint Property results in invalid constraint indices
Process 32481 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000164b3ea68 libTKFillet.7.7.2.dylib`ChFi3d_Builder::PerformIntersectionAtEnd(int) + 18772
libTKFillet.7.7.2.dylib`ChFi3d_Builder::PerformIntersectionAtEnd:
->  0x164b3ea68 <+18772>: ldr    x8, [x0]
    0x164b3ea6c <+18776>: ldr    x8, [x8, #0x90]
    0x164b3ea70 <+18780>: add    x1, sp, #0x4b0
    0x164b3ea74 <+18784>: blr    x8
Target 0: (freecad) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
  * frame #0: 0x0000000164b3ea68 libTKFillet.7.7.2.dylib`ChFi3d_Builder::PerformIntersectionAtEnd(int) + 18772
    frame #1: 0x0000000164b345fc libTKFillet.7.7.2.dylib`ChFi3d_Builder::PerformOneCorner(int, bool) + 5256
    frame #2: 0x0000000164aeb734 libTKFillet.7.7.2.dylib`ChFi3d_Builder::PerformFilletOnVertex(int) + 1056
    frame #3: 0x0000000164ae9a0c libTKFillet.7.7.2.dylib`ChFi3d_Builder::Compute() + 1404
    frame #4: 0x0000000164c48404 libTKFillet.7.7.2.dylib`BRepFilletAPI_MakeChamfer::Build(Message_ProgressRange const&) + 48
    frame #5: 0x0000000162bf0550 libTKTopAlgo.7.7.2.dylib`BRepBuilderAPI_MakeShape::Shape() + 84
    frame #6: 0x00000001633bebd4 Part.so`Part::TopoShape::makeElementShape(BRepBuilderAPI_MakeShape&, std::__1::vector<Part::TopoShape, std::__1::allocator<Part::TopoShape> > const&, char const*) + 256
    frame #7: 0x00000001633d1ea0 Part.so`Part::TopoShape::makeElementChamfer(Part::TopoShape const&, std::__1::vector<Part::TopoShape, std::__1::allocator<Part::TopoShape> > const&, Part::ChamferType, double, double, char const*, Part::Flip) + 1056
    frame #8: 0x0000000176fec298 _PartDesign.so`PartDesign::Chamfer::execute() + 736
    frame #9: 0x0000000102b15944 libFreeCADApp.dylib`App::DocumentObject::recompute() + 112
    frame #10: 0x0000000176f52088 _PartDesign.so`PartDesign::Feature::recompute() + 124
    frame #11: 0x0000000102a97104 libFreeCADApp.dylib`App::Document::_recomputeFeature(App::DocumentObject*) + 748
    frame #12: 0x0000000102a98ff0 libFreeCADApp.dylib`App::Document::recomputeFeature(App::DocumentObject*, bool) + 232
    frame #13: 0x000000029996e8b4 PartDesignGui.so`PartDesignGui::TaskChamferParameters::onSizeChanged(double) + 80
    frame #14: 0x00000001039b9e84 libQt5Core.5.dylib`___lldb_unnamed_symbol16969 + 820
    frame #15: 0x0000000101ad0230 libFreeCADGui.dylib`Gui::QuantitySpinBox::updateFromCache(bool, bool) + 272
    frame #16: 0x0000000101ace2c8 libFreeCADGui.dylib`Gui::QuantitySpinBox::validateInput() + 116
    frame #17: 0x0000000101ad17ec libFreeCADGui.dylib`Gui::QuantitySpinBox::focusOutEvent(QFocusEvent*) + 52
    frame #18: 0x0000000100feb758 libQt5Widgets.5.dylib`QWidget::event(QEvent*) + 104
    frame #19: 0x00000001010c1cf4 libQt5Widgets.5.dylib`QAbstractSpinBox::event(QEvent*) + 144
    frame #20: 0x0000000100fc6264 libQt5Widgets.5.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 232
    frame #21: 0x0000000100fc709c libQt5Widgets.5.dylib`QApplication::notify(QObject*, QEvent*) + 472
    frame #22: 0x0000000101661b58 libFreeCADGui.dylib`Gui::GUIApplication::notify(QObject*, QEvent*) + 204
    frame #23: 0x000000010399d094 libQt5Core.5.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
    frame #24: 0x0000000100fc4c98 libQt5Widgets.5.dylib`QApplicationPrivate::setFocusWidget(QWidget*, Qt::FocusReason) + 204
    frame #25: 0x0000000100fe8cb8 libQt5Widgets.5.dylib`QWidget::setFocus(Qt::FocusReason) + 392
    frame #26: 0x0000000100fe90b4 libQt5Widgets.5.dylib`QWidget::focusNextPrevChild(bool) + 240
    frame #27: 0x00000001010bd5e0 libQt5Widgets.5.dylib`QScrollArea::focusNextPrevChild(bool) + 16
    frame #28: 0x0000000100febfc4 libQt5Widgets.5.dylib`QWidget::event(QEvent*) + 2260
    frame #29: 0x00000001010c1cf4 libQt5Widgets.5.dylib`QAbstractSpinBox::event(QEvent*) + 144
[... cut ...]
    frame #50: 0x0000000190c32fa0 AppKit`-[NSApplication run] + 500
    frame #51: 0x000000010702c4d0 libqcocoa.dylib`___lldb_unnamed_symbol3684 + 968
    frame #52: 0x000000010399afa8 libQt5Core.5.dylib`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 312
    frame #53: 0x000000010399d578 libQt5Core.5.dylib`QCoreApplication::exec() + 120
    frame #54: 0x0000000101598104 libFreeCADGui.dylib`Gui::Application::runApplication() + 2300
    frame #55: 0x000000010000729c freecad`main + 3356
    frame #56: 0x000000018d5e3f28 dyld`start + 2236
(lldb) 
```

wwmayer, 2024-07-03:
The reason for the crash is in ComplexGeoData::setElementName. The member variable `_elementMap` is set to a null pointer at the point when calling it.
Here ComplexGeoData::setElementName must be fixed to throw an exception and do not let it crash.

Then SketchObject::buildShape() must also be changed to make sure the TopoShape has a valid element map. Calling `vertex.setElementMap({});` might be sufficient.

wwmayer, 2024-07-03:
With the above change and the changes of https://github.com/FreeCAD/FreeCAD/pull/15060/files it's possible to load and recompute the project.

Now when following the steps to update the chamfer I can reproduce the crash. This, however, looks like a bug in OCC.

FEA-eng, 2024-07-10:
@maxwxyz Shouldn't this be a blocker issue (especially considering the number of reports)? It seems to be a regression. From one of the duplicates (#15336):

> Weekly build 37819 does not crash, weekly build 38079 does.

wwmayer, 2024-07-13:
@bgbsww is there a fix for the crash inside `Data::ElementMap::setElementName` in sight? There were many bug reports about this crash in the last couple of days?

bgbsww, 2024-07-15:
Apologies, I didn't realize there was still work expected on this one.

There is this idiom now in many places in the code that I think explains a bit of what has happened here:

                    // Originally in ComplexGeoData::setElementName
                    // LinkStable/src/App/ComplexGeoData.cpp#L1631
                    // No longer possible after map separated in ElementMap.cpp

                    if (!elementMap()) {
                        resetElementMap(std::make_shared<Data::ElementMap>());
                    }

                    elementMap()->encodeElementName(indexedName[0], newName, ss, &sids, Tag, op);
                    elementMap()->setElementName(indexedName, newName, Tag, &sids);

So it seems I need to take a pass looking at all calls to setElementName and making sure that there is always an elementMap defined.

Or alternately revisit the decision to move this method; that predates me, so I can't answer the why, unfortunately.

wwmayer, 2024-07-15:
> There is this idiom now in many places in the code that I think explains a bit of what has happened here:

There are many duplications of almost identical code. Shouldn't the code be refactored to avoid them?

nazar-pc, 2024-07-20:
Just in case it helps with prioritization, I'm stuck at old dev build because new builds are literally unusable until this is fixed.

bgbsww, 2024-07-20:
Do you have an issue that wasn't fixed with #15456?  This is still here because of the refactor request, not the crash.

nazar-pc, 2024-07-20:
No crash in latest dev build, thanks! I was not aware of the crash fix.

bgbsww, 2024-07-23:
@maxwxyz All evidence appears to be that the crash is gone.  Is this still a blocker?

maxwxyz, 2024-07-23:
so could this issue then be closed completely or is there something left to do?

wwmayer, 2024-07-23:
There are two different crashes. One of it is fixed but not the other one. However, the second one is an OCC bug that we have no control over.

bgbsww, 2024-07-29:
I'm un-assigning since there are no remaining TNP issues here, and the refactor recommendation now has a PR to address it.  The OCC bug is out of my scope.  Let me know if there's something else.

drwho495, 2025-07-08:
im seeing something similar to this issue, im not exactly sure if its the same segfault tho.

i was able to fix it by adding a check, seeing if the elementmap is a nullptr in setElementName, and if it is then I return {}

i dont know enough about c++ to assign an empty map, especially with how TopoShapeExpansion is setup
here is a screenshot of the changes
[image]

pieterhijma, 2025-08-31:
@PaddleStroke, could you please take a look?

pieterhijma, 2025-09-15:
Thanks @PaddleStroke and @drwho495. I'm looking at the file I succeeded in somehow recovering the file. The original file has this as contents:

```xml
        <Object name="Fillet" Extensions="True">
            <Extensions Count="1">
                <Extension type="App::SuppressibleExtension" name="SuppressibleExtension">
                </Extension>
            </Extensions>
            <Properties Count="15" TransientCount="1">
                <_Property name="_Body" type="App::PropertyLinkHidden" status="251658241"/>
                <Property name="AddSubShape" type="Part::PropertyPartShape" status="128">
                    <Part file="PartShape13.brp"/>
                </Property>
                <Property name="Base" type="App::PropertyLinkSub">
                    <LinkSub value="Pocket002" count="2">
                        <Sub value="Edge27" shadow=";g85181;SKT;:He64,E;:G;XTR;:He64:7,F;:U;XTR;:He64:7,E;:He65,E;:He67,E;:He69,E.Edge27"/>
                        <Sub value="Edge22" shadow=";g85181;SKT;:He64,E;:He65,E;:He67,E;:He69,E.Edge22"/>
                    </LinkSub>
                </Property>
                <Property name="BaseFeature" type="App::PropertyLink" status="8">
                    <Link value="Pocket002"/>
                </Property>
```

The repaired one has this:

```xml
        <Object name="Fillet" Extensions="True">
            <Extensions Count="2">
                <Extension type="App::SuppressibleExtension" name="SuppressibleExtension">
                </Extension>
                <Extension type="Part::PreviewExtension" name="PreviewExtension">
                </Extension>
            </Extensions>
            <Properties Count="16" TransientCount="3">
                <_Property name="PreviewShape" type="Part::PropertyPartShape" status="152"/>
                <_Property name="_Body" type="App::PropertyLinkHidden" status="251658240"/>
                <_Property name="_ElementMapVersion" type="App::PropertyString" status="234881024"/>
                <Property name="AddSubShape" type="Part::PropertyPartShape" status="128">
                    <Part ElementMap="0.15.70200.4" file="Fillet.AddSubShape.brp"/>
                    <ElementMap/>
                </Property>
                <Property name="Base" type="App::PropertyLinkSub">
                    <LinkSub value="Pocket002" count="2">
                        <Sub value="Edge4" shadow=";#124:1c;:He69,E.Edge4"/>
                        <Sub value="Edge28" shadow=";#126:1c;:He69,E.Edge28"/>
                    </LinkSub>
                </Property>
                <Property name="BaseFeature" type="App::PropertyLink" status="8">
                    <Link value="Pocket002"/>
                </Property>
```

It appears to me that `try3.FCStd` was created with a development version where TPN wasn't yet stable. Can you confirm @drwho495?

There are similar differences in the other fillets.

pieterhijma, 2025-09-15:
Ok, great, thanks. This is the file where TPN is not an issue anymore:

[try3_redone.FCStd.zip](https://github.com/user-attachments/files/22346048/try3_redone.FCStd.zip)

pieterhijma, 2025-09-17:
Minimal model for this bug (remove the .zip extension):

[try3_redone-7-without-long-part.FCStd.zip](https://github.com/user-attachments/files/22383774/try3_redone-7-without-long-part.FCStd.zip)

pieterhijma, 2025-09-17:
Draw test harnesses for this bug that can be submitted to OCCT:

[chamfer-approaching-ellipse.zip](https://github.com/user-attachments/files/22384329/chamfer-approaching-ellipse.zip)

It is also problematic for fillets:

[fillet-approaching-ellipse.zip](https://github.com/user-attachments/files/22384334/fillet-approaching-ellipse.zip)

pieterhijma, 2025-10-29:
OCCT has fixed the crash. I can confirm that with the minimal model above, the bug doesn't appear anymore with the latest OCCT that incorporates the fix. In the preview you can see the chamfer and fillet appear and when you push beyond the accepted radius of 3.7 or 3.8, the preview shows that the chamfer/fillet will not produce a result (the preview has a red border). Creating a fillet or chamfer with these radii will create a broken one and FreeCAD will no longer crash.

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.1.0dev.44073 (Git)
Build date: 2025/10/28 20:29:01
Build type: Unknown
Branch: ir
Hash: 5b2987bd0751d5ee7ab95535367807fce54e803a
Python 3.13.7, Qt 6.10.0, Coin 4.0.5, Vtk 9.5.2, boost 1_89, Eigen3 5.0.0, PySide 6.10.0
shiboken 6.10.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 8.0.0.rc2-cd3fc87130
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * Ondsel-Lens 2024.10.21.01
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
```

I will mark this issue as Done in the OCCT Liaison project. Since FreeCAD won't ship with OCCT 8, we can keep this issue open until FreeCAD has moved to OCCT 8 or we can close it now.