FreeCAD/FreeCAD #15103: TechDraw: Snapping vertical dimensions - text is not centered

maxwxyz, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Dimension text of vertical dimensions is snapping but not centered:
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-04:
@PaddleStroke any news on this?

benj5378, 2024-08-25:
How should over under tolerance labels be handled? 

[image]

maxwxyz, 2024-08-25:
@benj5378 should be considered, so not like above image but centered overall
@pierreporte FYI

pierreporte, 2024-08-25:
@benj5378 The below right one, written vertically, is the right one. The whole content of the text (including prefixes, suffixes, and everything else) should be centered, not just the value. Note that the arrow length should be the same on both sides in this case, so strangely enough they look better on the above left.

benj5378, 2024-08-25:
> Note that the arrow length should be the same on both sides in this case, so strangely enough they look better on the above left.

Thanks for the feedback. The arrow cut off is due to screenshot cropping ;)

benj5378, 2024-08-28:
@PaddleStroke could you please help explain what is going on here? I get the first couple of lines with p1, p2, mid, normal, toCenter, but the posV and pos throws me off completely, and the following logic. Thanks.

https://github.com/FreeCAD/FreeCAD/blob/73cb5bafe0c32c178d9b594a6ca03124483d305f/src/Mod/TechDraw/Gui/QGIViewDimension.cpp#L187-L197

PaddleStroke, 2024-08-28:
pos is the mouse position of the label. But the origin of the label, which is around one of its corner.
posV is a vector2d of pos + toCenter. So basically the position of the center of the label.

projPnt is a projection of posV onto the normal (passing by mid)

Then you check if projPnt and posV are close enough and if so you update the value of pos, which will be used to update the position of the label.

benj5378, 2024-08-28:
@PaddleStroke Thanks, this helped a lot. I don't hope you mind if I rename the variable, and split up the functions

PaddleStroke, 2024-08-28:
No problems of course

WandererFan, 2024-09-02:
This LGTM.   No crashes or unexpected behaviour.  Thanks @benj5378.

I'll leave it to the experts to raise a new issue on how to treat the blue dim in the picture.
[image]

PR #16060 (2024-09-02): [TechDraw] Make dimension labels centered
Partially fixes #15103

I have no idea, why the bounding rect was offset with 4 times linewidth. Might need to check if removing this has any side effects.
commit 8389689b (2024-09-02): [TechDraw] Fix incorrect centering of dimension label

and improve rendering of frame
Fixes #15103
closed 2024-09-02 by WandererFan