FreeCAD/FreeCAD #15108: "Settings / General / Import profile" ok, but export not possible

EmEiBee, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It doesn't seem possible to export all settings in order to provide the same working environment on a second FC instance.
Under "Settings / General / Import profile" there is a "Save new" button, but it only generates an entry in the field above, not an export file that I could import into the second FC instance.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Royalty252, 2024-07-25:
Hi, I would like to contribute to this issue as part of the FOSS hackathon. I hope you find this request.

Chin2202, 2024-07-27:
i would liketo contribuite to this issue

luzpaz, 2024-07-27:
Please present your proposals

maxwxyz, 2024-12-13:
Fixed by https://github.com/FreeCAD/FreeCAD/pull/15235

closed 2024-12-13 by maxwxyz