FreeCAD/FreeCAD #15110: Start: recent images do not appear in Recent Files widget

furgo16, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Start FreeCAD
2. Load an image file with a supported format (e.g. `jpg`) via the File > Open... menu
3. Check that the image is loaded in the document => OK
4. Check that the image's file name appears in the File > Open Recent dropdown => OK
5. Check that the image thumbnail appears as a card in the Start page's Recent Files => FAIL
6. Restart FreeCAD
1. Check that the image's file name appears in the File > Open Recent dropdown => OK
1. Check that the image thumbnail appears as a card in the Start page's Recent Files => FAIL
1. **Expected**:  the image appears as a card in the Start Page's Recent Files widget

This used to work with 0.21 but it no longer works with 0.22 with the rewritten Start page

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37943 (Git) Snap 1063
Build type: Release
Branch: main
Hash: 554310a6ba9217cf3aaf2b301772d8b4ce50db9f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-02:
When did you save the file?

furgo16, 2024-07-02:
> When did you save the file?

I did not save the file. I just opened it as per the steps to reproduce.

maxwxyz, 2024-07-02:
I cannot reproduce, it works for me on:

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.26.2
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21

```

tomate44, 2024-07-09:
I cannot reproduce either. File and thumbnail appear on Start Page at step 8.
```
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38090 (Git)
Build type: Unknown
Branch: main
Hash: fa2741fd292909eebcad965ebcdf3599bd20f25b
Python 3.12.4, Qt 6.7.1, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * casat
  * sheetmetal 0.4.3
  * OpenTheme 2024.4.20
  * CurvesWB 0.6.40

```

furgo16, 2024-07-10:
Further discussion at: https://forum.freecad.org/viewtopic.php?t=89018

Looking at the code, a few things stand out:

1. The image file is correctly loaded in `RecentFilesModel` (effectively the list of recent files in the menu), but not on `DisplayedFilesModel` (the thumbnails on the Start page). That's what seemingly is causing this issue.
2. This is due to the extra `freecadCanOpen()` check that happens before adding the file to `DisplayedFilesModel`. At the time this check is made, the image formats have not yet been added to the list of supported formats (thanks @Syres916 for pointing that out). So the image is deemed as not supported, and not loaded.
   https://github.com/FreeCAD/FreeCAD/blob/52b4112a9a5fd6dcb43b69839cdd9def4c637230/src/Mod/Start/App/DisplayedFilesModel.cpp#L187-L203
1. That check seems superfluous. If the image file is already loaded in the `RecentFilesModel`, it must have been checked to be supported already. Why the second check?
1. Expanding on the above, removing the check below fixes the issue for me:
   https://github.com/FreeCAD/FreeCAD/blob/52b4112a9a5fd6dcb43b69839cdd9def4c637230/src/Mod/Start/App/DisplayedFilesModel.cpp#L193-L195

Finally, on the second point, I'm not sure if it's a race condition, or if the console output can be trusted to be sequential (I added a few debug warnings on the console to test), but for me, `setImportImageFormats()` seems to be called after `DisplayedFilesModel::addFile()` is executed. If that's the case, that would indeed explain why the image formats cannot be loaded.

But it would not explain why it works for some people and not others.

mfraser, 2024-07-29:
I'm seeing this with FreeCAD installed from the daily PPA, but not from the latest weekly AppImage and it has been like this for a while.
Wasn't going to report it as there were changes going on with the Start page and I thought it was related until I ran the AppImage yesterday trying to fix something else and the thumbnails were visible.

maxwxyz, 2024-11-03:
@furgo16 is this still an issue?

furgo16, 2024-11-03:
@maxwxyz yes, thanks for checking. The same for the main branch.

furgo16, 2024-11-03:
Perhaps a developer could have a quick look at the findings on https://github.com/FreeCAD/FreeCAD/issues/15110#issuecomment-2221110812? Thanks.

furgo16, 2024-12-03:
As of today, I can no longer reproduce the issue. I'd leave it open though, as I don't know what fixed it and a few questions remain unanswered. Thanks. See also https://github.com/FreeCAD/FreeCAD/issues/13676#issuecomment-2514419195

closed 2025-02-18 by furgo16