FreeCAD/FreeCAD #15113: Start page: missing vertical scrollbars for Recent Files and Examples sections

furgo16, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On the Start page, both the Recent Files and Examples sections are widgets that contain a number of thumbnail cards arranged in rows. The vertical size of the widget is fixed, but the number of cards can encompass multiple rows, which are then hidden from view.

The only way to access those initially hidden rows is to hover onto the widget and scroll with the mouse wheel (or other means depending on the user setup). However, there is no visual indication that a) the rows are scrollable and b) which scrolling position we're at.

Either the widget should autoexpand its height to encompass all thumbnail rows, or it should have a vertical scrollbar.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37943 (Git) Snap 1063
Build type: Release
Branch: main
Hash: 554310a6ba9217cf3aaf2b301772d8b4ce50db9f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2025-02-26:
This issue was never fixed for me. I just now realize it's been closed. I've tested it again with the FreeCAD Classic theme and in safe mode. Still no scroll bars for the Start page widgets.

@maxwxyz would it be possible to reopen it? Thanks.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.40419 (Git) Snap 1314
Build type: Release
Branch: main
Hash: cf477695f9662d9892d5fc79e83065b4ac3d52bf
Python 3.10.12, Qt 5.15.15, Coin 4.0.0, Vtk 7.1.1, IfcOpenShell 0.8.0, OCC 7.7.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Logical/physical DPI: 96/95.1083
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.37
  * fcVM
  * ezydxf
  * Arch
  * Assembly4 0.50.17
```

maxwxyz, 2025-02-26:
Would it be better to have more width and therefore scroll only horizontally instead of two independent vertical scroll areas?
@chennes FYI

hyarion, 2025-02-27:
When reading the source code, it looks like this is intentional to always hide scroll bars: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Start/Gui/FileCardView.cpp#L41-L42

But for this to work, the filecardview needs to resize with the content.

This can be tested by resizing the start view to the minimal horizontal size.

furgo16, 2025-02-27:
> This can be tested by resizing the start view to the minimal horizontal size.

This does not work unfortunately. That's exactly how I test it for widgets that have only one row (e.g. Examples)

furgo16, 2025-02-27:
It would be fine to hide the vertical scrollbars when they are not needed (i.e. all visible content is in the file card), but it's very disconcerting when the file card does not show all content and suddenly when you're scrolling the page (not the file card), new content pops up on the filecard. It seems for others "pops up" is actually auto scrolling. In my case it seems to be in discrete steps.

furgo16, 2025-02-27:
On the first part, I'm just moving the scroll wheel up and down. Notice:

- Both the Start page and the file card are scrolling at the same time
- There is no smooth scrolling on the file card. One row of content is replaced by another

On the second part, I'm dragging the scroll bar with the mouse cursor. Then only the Start page is scrolling.

https://github.com/user-attachments/assets/eb5d6ff2-5b94-41e0-9537-ed89cb044d05

hyarion, 2025-02-27:
> > This can be tested by resizing the start view to the minimal horizontal size.
> 
> This does not work unfortunately. That's exactly how I test it for widgets that have only one row (e.g. Examples)

I was probably not clear enough. I mean this issue can be reproduced by resizing the start view to the minimal horizontal size.

There shouldn't be scroll bars in this part of the view. The examples area needs to resize to fit all items (ie if scrolling is needed, then it needs to be made larger vertically). Allowing scrolling just makes it weird.

furgo16, 2025-02-27:
> I was probably not clear enough. I mean this issue can be reproduced by resizing the start view to the minimal horizontal size.

I think we're saying the same thing.

> There shouldn't be scroll bars in this part of the view. The examples area needs to resize to fit all items (ie if scrolling is needed, then it needs to be made larger vertically). Allowing scrolling just makes it weird.

I fully agree.

furgo16, 2025-02-27:
So I think in summary, this issue could be reworded: file cards do not vertically resize to fit all contents.

If a Qt expert can give me some hints on where to start looking at, I would try to have a go at a fix.

commit db3358d5 (2024-08-12): Make start page use QStackedWidget + other fixes

closes #14039, closes #14694, closes #14817, closes #15113
PR #19932 (2025-03-05): Start: fix fileview use scrollbars as needed
if for whatever reason the viewport failed to resize let qt add the scrollbars

fixes #15113

if #19901 gets merged this will be very unlikely to happen, to my understanding it's an issue caused by lag between render and viewport realization, with the utilization of painter and styled delegate i couldn't reproduce the issue anymore, still this should be added as safeguard in case it happens to someone
closed 2025-03-05 by chennes