FreeCAD/FreeCAD #15117: Sketch: argument not found

szimszon, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open 2a_fancy FreeCAD file
[image]
The sketch is broken with:

```
14:24:32  9e-08 <App> Document.cpp(3152): exception in _a_fancy#Sketch thrown: argument not found
14:24:32  Sketch: argument not found
14:24:53  20.8493 <App> Document.cpp(3152): exception in _a_fancy#Sketch thrown: argument not found
14:24:53  Sketch: argument not found
14:24:53  20.935 <App> Document.cpp(3152): exception in _a_fancy#Sketch thrown: argument not found
14:24:53  Sketch: argument not found
```
It has just one external dependency from a BaseFeature which is from a Binder originating in another document **2a_corner**.

It doesn't make any difference if I make a Link instead of a subshape binder.

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Hungarian/Hungary (hu_HU)
Installed mods: 
  * Defeaturing 1.2.2
  * lattice2 1.0.0
  * Curves 0.6.40
  * Ondsel-Lens 2024.5.6.01
  * fasteners 0.5.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

szimszon, 2024-07-02:
[2a.zip](https://github.com/user-attachments/files/16068178/2a.zip)

wwmayer, 2024-07-02:
https://github.com/FreeCAD/FreeCAD/pull/15060

closed 2024-07-04 by wwmayer