FreeCAD/FreeCAD #15120: Quickmeasure displays information about the wrong geometry in sketcher

Ely31, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this file while editing the sketch, I select the line on the right side and quickmeasure gives me a radius value. I select the top arc and it gives me a distance value. All the other geometry in the sketch measures correctly.

### Full version info

```shell
OS: KDE neon 6.0 (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Unknown
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.10.12, Qt 5.15.13, Coin 4.0.0, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * IconThemes
  * fasteners 0.5.24
  * OpenDark 2024.3.13
  * OpenTheme 2024.6.22
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/78882466/b40b37f1-404a-4644-b70d-e91f202229ad
[breaks quickmeasure.zip](https://github.com/user-attachments/files/16069902/breaks.quickmeasure.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-02:
@PaddleStroke FYI, can confirm

PaddleStroke, 2024-07-03:
So it seems that the selection in edit mode is wrong.
See in video below :
https://github.com/FreeCAD/FreeCAD/assets/19984177/07f3fadf-f093-440f-a914-e910869f2c90
[sketcher edge switching.FCStd.zip](https://github.com/user-attachments/files/16081219/sketcher.edge.switching.FCStd.zip)

the arc outside of edit mode is 'sketch.edge1' then inside edit mode it becomes 'sketch.edge2'

PaddleStroke, 2024-07-03:
Ok so I think it is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/7657

Quickmeasure is just a symptom of this issue.

maxwxyz, 2024-07-03:
understood. Is it easier to fix #7657 because of TNP fixes or if not, make a temporary workaround for this issue?

PaddleStroke, 2024-07-03:
So the problem actually runs deeper. Sketch.edges is not updated until you leave edit mode. A short video will make it easier to explain : 

https://github.com/FreeCAD/FreeCAD/assets/19984177/5a3d06ff-3b22-4e9d-a0f0-a85ff7695578

So hmm. Either quickmeasure can handle sketch edit mode specifically. Else we better disable quickmeasure while in edit mode.

closed 2024-07-15 by chennes