FreeCAD/FreeCAD #15126: Dark & litght theme: drop-down menus don’t have a shadow

pierreporte, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the new official dark and light themes, drop-down menus no longer have shadows. They are harder to spot because they don’t contrast with background anymore.

[image]
[image]

Expected result:

[image]

### Full version info

```shell
OS: Ubuntu 23.10 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

### Subproject(s) affected?

None

### Anything else?

Original issue: #14745

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-07-02:
@maxwxyz @MisterMakerNL

MisterMakerNL, 2024-07-02:
I'll add a border but the shadow seems to be done by Qstyle, and so far I have not found a way to modify it.
It\s just there in windows.

MisterMakerNL, 2024-07-02:
Added borders:
https://github.com/FreeCAD/FreeCAD/pull/14749/commits/c6197c75aafff378ee5e3569b80606cf73e7fd8b

luzpaz, 2024-07-27:
What's left here ?

maxwxyz, 2024-07-27:
Should be fixed with https://github.com/FreeCAD/FreeCAD/pull/14749
@pierreporte please reopen if there still is an issue.

closed 2024-07-27 by maxwxyz

pierreporte, 2024-07-28:
The border issue is fixed but there is still a contrast difference between light theme and the two others. I don’t think it qualifies for reopening but I couldn’t do it anyway.