FreeCAD/FreeCAD #15130: Revolve loses axis, adding it back results in "AxisLink shape is not an edge"

theosib, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached file, please find the revolve named "wheel snap revolve". Somehow it lost its axis and threw a naming error, so I decided to try to repair it by clicking the ellipsis on its Axis Link and selecting the correct sketch, which is called "wheel snap axis". It now says "AxisLink shape is not an edge".

Technically, the axis is a sketch, but (a) the only thing in the sketch is a line segment and some positional constraints, and (b) there is no means by which to select something inside a sketch when setting the Axis Link.

I asked for help with this on Discord, and I'm doing the right thing. I also searched the issue tracker, and I didn't find a report on this. So I decided to report it.

[car.FCStd.zip](https://github.com/user-attachments/files/16072968/car.FCStd.zip)

Thanks.

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-07-02:
I figured it out. Since the only thing in the sketch was one line, there was no way for FreeCAD to figure out if I was selecting the sketch of part of the sketch. So I added an extra useless line to the sketch, and then I was able to go back and select the just the edge I wanted for revolving.

So the bug is that if your axis for revolving is a sketch with only a single line element in it, FreeCAD will select the whole sketch rather than the edge it wants.

One possible solution is for FreeCAD to realize that the sketch has only one line in it and automatically choose that rather than linking to the whole sketch.

FlachyJoe, 2024-07-02:
If you select the edge in the 3D view it works as expected.

theosib, 2024-07-02:
> If you select the edge in the 3D view it works as expected.

You mean like here? Because that was the first thing I tried, and I did try multiple times. it always selects the whole sketch, not the edge inside the sketch.

[image]

FlachyJoe, 2024-07-02:
Please try with the latest dev.

theosib, 2024-07-02:
> Please try with the latest dev.

You mean 38001?

Same problem. I click the ellipsis, select the sketch in the 3D view, and press tab or click out of the constraints row. Only the whole sketch remains selected. There is no way to select just part of the sketch.

theosib, 2024-07-02:
Ok, I think I figured it out properly this time.

Evidently when you click the 3D view, the Link window doesn't close. It just get put behind the main window in the stacking order. Since the Link window stays open, it interferes until you do something to accidentally close it like clicking on another property.

I don't think most user are going to realize that the window is still open and just hidden behind the main one. At least not any way other than accidentally.

I think the solution is to set the window so that it stays on top.

theosib, 2024-07-03:
I don't know if it makes a difference, but I was using the Part workbench, not Part Design.

luzpaz, 2024-08-01:
Is this specific for macos ?

FEA-eng, 2025-10-19:
Still relevant

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```