FreeCAD/FreeCAD #15131: New dark theme: Invisible objects in the tree view have black text immediately after startup

dapetcu21, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Immediately after starting FreeCAD, invisible objects in the tree look like this:

Selecting each of them then pressing spacebar twice (to toggle visibility on then off again) makes them render the correct color (a light grey).

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-02:
cannot reproduce this, can you maybe add the freecad file. Although I dount it will help we can try.

dapetcu21, 2024-07-02:
Sure. It may be macOS-specific?

[box-2x4.FCStd.zip](https://github.com/user-attachments/files/16074155/box-2x4.FCStd.zip)

dapetcu21, 2024-07-02:
Light theme doesn't have this issue and neither does Classic **when the OS is in Light Mode**. With macOS in Dark Mode, Classic has the same problem:

[image]

depthoffocus, 2024-07-02:
I have experienced this issue.

It happens with the new dark theme in the latest weekly (38001) on macOS x86-64 **with the start page disabled** if you start FreeCAD by double-clicking on a FreeCAD file. If the start page is enabled, the problem does not occur.

Even with the start page disabled, if you launch FreeCAD _then_ double-click the file (or load it, or drag it in) the problem does not occur. Nor does it occur if you launch FreeCAD and then open a new document.

depthoffocus, 2024-07-03:
Also this affects the latest OpenDark in the same way.

MisterMakerNL, 2024-07-03:
I suspect an overlay issue here. Wonder what happens if you switch to another overlay like light/light or dark/light. This might be the same issue withthe headers not picking up formatting.

luzpaz, 2024-08-30:
> I suspect an overlay issue here. Wonder what happens if you switch to another overlay like light/light or dark/light. This might be the same issue with the headers not picking up formatting.

ping @dapetcu21 @depthoffocus

depthoffocus, 2024-08-30:
> > I suspect an overlay issue here. Wonder what happens if you switch to another overlay like light/light or dark/light. This might be the same issue with the headers not picking up formatting.
> 
> ping @dapetcu21 @depthoffocus

I am not wholly sure what you mean, but eventually I found the revelant preference. if I select for example Dark Theme and Light Background in **Preferences > Display > UI > Overlay Stylesheet**, it is no different?

I am not currently keeping up very well with developer releases at the moment due to other time commitments; but it's still a problem in 38553.

luzpaz, 2024-09-20:
@johnocfii could you possibly reproduce this and see if the workaround fixes it ?

JohnOCFII, 2024-09-21:
> @JohnOCFII could you possibly reproduce this and see if the workaround fixes it ?

I tried to recreate this issue using both `FreeCAD Dark` and `OpenDark` and was unable to experience the issue of objects appearing invisible due to dark text on a dark background. I tried both with the overlay on and the overlay off.

[image]

[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git)
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

depthoffocus, 2024-09-22:
What is the "workaround"? Changing overlays had no effect for me. The issue still occurs on 38794 on the Mac. 

- If FreeCAD is not running
- you double-click an item in the finder, so causing FreeCAD to launch
- some of the treeview items are black-on-dark in FreeCAD Dark.

Setting a different overlay doesn't seem to have helped.

(As an aside, changing themes while a document is open now hangs FreeCAD)

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47

JohnOCFII, 2024-09-22:
After reading the most recent items, I tried again, and realized I had missed two critical items for this to appear:

- macOS set to dark mode at the OS level (mine is normally set to follow the sun, so was in daytime mode when I tested yesterday)
- Launch FreeCAD by double-clicking a FreeCAD document, and not by launching FreeCAD App first

With those two procedures changes, I too see the issue.

I did not have the issue of FreeCAD crashing when changing themes with a document open.

To me, the simplest workaround was to launch FreeCAD app first, then open the document from within FreeCAD.
Also - the problem seemed to occur much less frequently in my testing with OpenDark vs. FreeCAD dark.  In either case, it did not happen every time.  Some odd interactions between how the App is launched and theming initiation.

Also -- switching between OpenDark, FreeCAD Dark, OpenLight and FreeCAD Light takes about 6-7 seconds on my M1 Mac.  I know this was called out weeks ago, and sped up somehow -- it seems the performance issue is back.

I have to wonder if a corrupt `user.cfg` file might be in play here.  Might be worth renaming it temporarily as a test.
On macOS, the files will be located in: `/Users/<YOURUSERNAME>/Library/Preferences/FreeCAD/ 

```
OS: macOS 15.0
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git)
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.29
  * Curves 0.6.47
```

[image]

depthoffocus, 2024-09-22:
FWIW I don't have macOS switched to Dark Mode -- I have it set to Light mode,

MisterMakerNL, 2024-09-22:
The lag was caused by setting Qwidget, I removed Qwidget from the stylesheet, I am not sure if @obelisk79 also removed it from Opentheme.
I see in the info that QTstyle is set too default QT which isn't correct.
Although I checked mine and it is ampty.
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38797 (Git)
Build type: Release
Branch: main
Hash: 86fa65ba489f0168145a1604423999b08090f857
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: Dutch/Netherlands (nl_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Assembly3 0.12.2
  * BIM021 2021.12.0
  * Dracula 0.0.5
  * fasteners 0.5.27
  * FeedsAndSpeeds 0.5.0
  * freecad_metal_workbench 0.0.1
  * Iconpack 2023.10.25
  * IconThemes (Disabled)
  * lattice2 1.0.0
  * OpenTheme 2024.8.17
  * OpenTheme.backup1720258469.8390777 2024.7.3 (Disabled)
  * OpenTheme.backup1722450713.1457298 2024.7.3 (Disabled)
  * Render 2024.8.4
  * scanObjects
  * sheetmetal 0.4.21

```

depthoffocus, 2024-11-11:
Has this been resolved? I am not seeing it now:

```
OS: macOS 14.7
Architecture: x86_64
Version: 1.1.0dev.39153 (Git)
Build type: Release
Branch: main
Hash: 3a7c8ed1c595c2a97116141589ed8f3fb0fd0bb5
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.92.0
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * lattice2 1.0.0
  * CurvedShapes 1.0.12
  * Curves 0.6.51
```

FEA-eng, 2025-10-02:
@maxwxyz Can you check if it's still relevant on Mac?

closed 2025-10-02 by maxwxyz