FreeCAD/FreeCAD #15132: Performance issues with new themes

maxwxyz, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We have several mentions of performance issues with the new included Light and Dark themes. I thought this was resolved already but the PR solved only the long loading times when switching themes.

This only affects the Light and Dark theme shipped with FreeCAD, the classic theme works normally. Also 3rd party themes like OpenTheme works normally.
I can reproduce this on Windows and it was reported also from users on Linux using the AppImage weekly (https://github.com/FreeCAD/FreeCAD/issues/15026).

FreeCAD is not usable, it lags overall. In this example I just try to open a Sketch in a simple document (one body, a pad and a sketch of two arcs connected by two lines). As soon as I move the mouse cursor I have clicked.
In normal (larger) documents I cannot navigate or select the features I want in the 3D view, because of the massive lag.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/72f6b8ab-cca1-4a72-9f47-32b885a16aa7

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.26.2
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-02:
@MisterMakerNL @kadet1090 @obelisk79 FYI

MisterMakerNL, 2024-07-02:
I see no difference between opentheme in the weekly and the default theme's in my pr I do not have lag issues there. But I do have them in the weekly.
Please merge this:
https://github.com/FreeCAD/FreeCAD/pull/14749

maxwxyz, 2024-07-21:
I have no more issues in 
```
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * CfdOF 1.27.4
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.22
  * Curves 0.6.41
```

so I'm closing this

closed 2024-07-21 by maxwxyz