FreeCAD/FreeCAD #15133: New dark theme: Overlay panel header is white text on a white background

dapetcu21, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-02:
Yeah I wanted to make an issue about this, I also get this. You need to reboot Freecad and it should be fixed.

dapetcu21, 2024-07-02:
Hmm, rebooting didn't do anything for me.

MisterMakerNL, 2024-07-07:
Was indeed not a restart:
But I fixed it here:
https://github.com/FreeCAD/FreeCAD/pull/14749

luzpaz, 2024-07-22:
Can we close this ticket ?

dapetcu21, 2024-07-24:
Yup. It's fixed!

closed 2024-07-24 by dapetcu21